MarlinFirmware/Marlin #17713: XZ YZ skew compensation Delta UBL incompatibility [BUG] (short description)

pediegoivan, 2020-04-25:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-04-26:
It may be that UBL is not compatible with the skew feature. What about ABL Bilinear? It is more fully integrated into the motion system.

pediegoivan, 2020-04-26:
I am really sorry but I do not know how to use ABL bilinear as it uses completely different G29 parameters.
I was running Marlin 2.0000
I am now updating and trying again with UBL.
I hope it works.
I might research how to use bilinear.

pediegoivan, 2020-04-26:
I am so sorry dude.
I placed my efforts on doing this properly. I did everything to update the marlin firmware from my printer before posting this issue.
I confirmed that it wasn't working, and then when I tried the bilinear variant, found the new .bin file in the sd card ... Never flashed.
Apparently I did not place the sd card propperly :S
I updated the firmware and now it works with UBL
Everything seems fine.

closed 2020-04-26 by pediegoivan