MarlinFirmware/Marlin #17716: [BUG] UBL Mesh Validation never stops priming

Grogyan, 2020-04-26:
### Bug Description

Running UBL Mesh Validation from the display, causes the extruder to never stop priming filament

### My Configurations
SKR1.3 + full graphic display + BLTouch

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4534393/Configuration.zip)

### Steps to Reproduce

1. Run Auto Home from the display
2. From the UBL Menu, run the step-by-step mesh validation

**Expected behavior:** 

1. The bed and nozzle heats up
2. Prime the nozzle 10mm 
3. Chirp once?
4. Stop prime and retract filament
5. Run the mesh validation routine

**Actual behavior:** 

1. The bed and nozzle heats up
2. Primes the nozzle 10mm
3. Chirp every 0.25mm extruded

4. Repeats the last 2 steps endlessly

#### Additional Information

https://www.youtube.com/watch?v=fpssDjqUrBA

#16287 is similar

phazei, 2020-04-27:
I tried a UBL test print.  It doesn't know how much to prime, so it will continually prime till you push the button.  Or at least that's what I presume and found out after it was priming for a bit.
If the screen said, "Press to continue when priming ready" or some variation that was short enough, it would be a lot more clear.

boelle, 2020-06-20:
@Grogyan is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle