MarlinFirmware/Marlin #17723: too few arguments to fuction serialprintPGM [BUG]

oriax1578, 2020-04-26:
### Bug Description
serial.h gives me an error while compiling  
`"too few arguments to function 'void serialprintPGM(const char*)'`
This is right after doing git pull and build. Also after complete fresh download of Marlin.

### My Configurations

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4534819/configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Download latest commit
2. Build Marlin
3. Error 

**Expected behavior:** For Marlin to compile without  errors

**Actual behavior:** Gives me error

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-26:
Issue confirmed, but only happens if  you define a X or Y Driver as a TMC stepper driver. So Configurations where required to identify this

ellensp, 2020-04-26:
This is very ugly.
But In configuration_store.cpp https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/configuration_store.cpp#L3220  change to
```
      /**
       * TMC stepper driver current
       */
      CONFIG_ECHO_HEADING("Stepper driver current:");

      #if AXIS_IS_TMC(X) || AXIS_IS_TMC(Y) || AXIS_IS_TMC(Z)
        say_M906(forReplay);
        SERIAL_ECHOLNPAIR_P(
          #if AXIS_IS_TMC(X)
            SP_X_STR, stepperX.getMilliamps()
          #endif
          #if AXIS_IS_TMC(X) && AXIS_IS_TMC(Y)
          ,
          #endif 
          #if AXIS_IS_TMC(Y)
            SP_Y_STR, stepperY.getMilliamps()
          #endif
          #if (AXIS_IS_TMC(X) || AXIS_IS_TMC(Y)) && AXIS_IS_TMC(Z)
          ,
          #endif 
          #if AXIS_IS_TMC(Z)
            SP_Z_STR, stepperZ.getMilliamps()
          #endif
        );
      #endif
```
But this is so ugly, I don't want to submit a PR request...  
Same issue with commas exists in following code in #if AXIS_IS_TMC(X2) || AXIS_IS_TMC(Y2) || AXIS_IS_TMC(Z2)

ellensp, 2020-04-26:
Fully patched file.
[configuration_store.zip](https://github.com/MarlinFirmware/Marlin/files/4535122/configuration_store.zip)

closed 2020-04-27 by thinkyhead