MarlinFirmware/Marlin #17724: [BUG] cannot compile bugfix branch.

hamster65, 2020-04-26:
Compiling .pio/build/LPC1768/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/eeprom_spi.cpp.o
Marlin/src/HAL/LPC1768/HAL_SPI.cpp:216:6: error: no declaration matches 'void SPIClass::beginTransaction(SPISettings)'
  216 | void SPIClass::beginTransaction(SPISettings cfg) {
      |      ^~~~~~~~
In file included from Marlin/src/HAL/LPC1768/HAL_SPI.cpp:52:
Marlin/src/HAL/LPC1768/include/SPI.h:42:10: note: candidate is: 'void SPIClass::beginTransaction(const SPISettings&)'
   42 |     void beginTransaction(const SPISettings&);
      |          ^~~~~~~~~~~~~~~~
Marlin/src/HAL/LPC1768/include/SPI.h:39:7: note: 'class SPIClass' defined here
   39 | class SPIClass {
      |       ^~~~~~~~
Compiling .pio/build/LPC1768/src/src/HAL/shared/esp_wifi.cpp.o
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/HAL_SPI.cpp.o] Error 1

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/4535040/conf.zip)

Evg33, 2020-04-26:
same problem for skr14t

Patag, 2020-04-26:
Same here on SKR v1.3

Powerprobot, 2020-04-26:
I confirm the Bug. Marlin didn't compile with SPI-tewak...
change in : ..\Marlin\src\HAL\LPC1768\include\SPI.h
in Row 42: from :  void beginTransaction(const SPISettings&);
to:   void beginTransaction(SPISettings);
after the change marlin will compile

@thinkyhead : please revert the commit  (tweak SPI) 
eebb68cd7f260d02a21e403b528547922203cbae

sjasonsmith, 2020-04-26:
No need to revert, it is easy to fix. I've posted a PR, although I'll understand if thinkyhead gets there first and just fixes it.

Powerprobot, 2020-04-26:
Nice! it works!

Evg33, 2020-04-26:
With #17734 works !
Thank you !

closed 2020-04-27 by thinkyhead

kloydeb, 2020-05-26:
> No need to revert, it is easy to fix. I've posted a PR, although I'll understand if thinkyhead gets there first and just fixes it.

what is a PR and how do i find it?

sjasonsmith, 2020-05-26:
> > No need to revert, it is easy to fix. I've posted a PR, although I'll understand if thinkyhead gets there first and just fixes it.
> 
> what is a PR and how do i find it?

@kloydeb this went I to the bugfix-2.0.x branch almost a month ago. If you scroll up in this issue you will see a note that says merged. That was the Pull Request (PR).

kloydeb, 2020-05-26:
> @kloydeb this went I to the bugfix-2.0.x branch almost a month ago. If you scroll up in this issue you will see a note that says merged. That was the Pull Request (PR).

does that mean it's fixed?

sjasonsmith, 2020-05-26:
> does that mean it's fixed?

Yes, but only in the bugfix-2.0.x branch. I don't know when the last versioned release occurred, but I don't think it is in it yet.