MarlinFirmware/Marlin #17728: [BUG] M42 not working

SuperBest27, 2020-04-26:
Hello, 

I tried using M42 command to change the pin mode and status. I tried to set pins to output mode with state TRUE. Setting the pin mode works great, however the state of the pin doesn't change, whatever I do. To check if the pins changed of mode and state, I used the M43 command. 

Steps to reproduce : 
Compile marlin 2.0.5.3 for an ender 3 and an skr mini e3 and enable the pins_debugging feature.
 
Thanks

ellensp, 2020-04-26:
What M42 command exactly?

SuperBest27, 2020-04-26:
> What M42 command exactly?

I used "M42 I M1 P46 S1"
M42 ignore fw protection set pin 46 to output with state 1

ellensp, 2020-04-26:
If you had to use  ignore fw protection, then the firmware is already using the pin.. the result is unpredictable. 
undefine what ever is using this pin. (see M43 to see what)  and run it without ignore
Secondly S1 is pwm 1, start with S255 and S0 ie on/off

ellensp, 2020-04-27:
I tied this on current bugfix and these tests worked as expected

After power on, status of the pin is"
PIN:  46 PC14         <unused/unknown>                       Input  = 1
Sent a "m42 p46 s255" the result is 
PIN:  46 PC14         <unused/unknown>                       Output = 1
Sent a "m42 p46 s0" the result is 
PIN:  46 PC14         <unused/unknown>                       Output = 0
Sent a "m42 p46 s1" the result is 
PIN:  46 PC14         <unused/unknown>                       Output = 1

All working as expected
I added in the I option (even though it isn't needed) no change, still works

I added in the M1, the commend didn't do anything.
I attempted to set M0, the command didn't do anything.

It seems like there is an issue with the Mode settings command option.

ellensp, 2020-04-27:
Bug is obvious... https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/control/M42.cpp#L51
```
  if (parser.seenval('M')) {
    switch (parser.value_byte()) {
      case 0: pinMode(pin, INPUT); break;
      case 1: pinMode(pin, OUTPUT); break;
      case 2: pinMode(pin, INPUT_PULLUP); break;
      #ifdef INPUT_PULLDOWN
        case 3: pinMode(pin, INPUT_PULLDOWN); break;
      #endif
      default: SERIAL_ECHOLNPGM("Invalid Pin Mode");
    }
    return;    //<---  this return should not be. it stops the s parameter being processed.
  }
```

closed 2020-04-27 by thinkyhead

SuperBest27, 2020-04-27:
> Bug is obvious... https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/control/M42.cpp#L51
>
> ```
>   if (parser.seenval('M')) {
>     switch (parser.value_byte()) {
>       case 0: [...]

I got to say that I'm not an expert of marlin code, but if I understood everything, just removing the return line will solve the issue?
I'm sorry I'm not very good at this :(

ellensp, 2020-04-27:
Yes that is correct. and it has been applied to marlins going forward.

SuperBest27, 2020-04-27:
Well thank you very much, I'll try this soon and I'll tell you if that worked.