MarlinFirmware/Marlin #17729: [FR] DELTA_CALIBRATION_RADIUS & MIN_PROBE_EDGE compatibility

pediegoivan, 2020-04-26:
### Description

I updated my firmware from 20000 and found that DELTA_CALIBRATION_RADIUS is not available anymore.
With the 2.0.5.3 build, you have only one parameter to select both, the area for the mesh leveling and the G33 calibration radius.
With older versions, you could manage to get different mesh probing radius and delta calibration radius.

### Feature Workflow

1. Set DELTA_CALIBRATION_RADIUS (to select the calibration radius for G33)
2. Set MIN_PROBE_EDGE (to select the mesh leveling area for G29)

sjasonsmith, 2020-04-26:
Can you explain why you need an explicitly set calibration radius? It auto selects the largest circle it can which is concentric to the bed. It factors in probe offset and MIN_PROBE_EDGE to do so.

Probing anything larger than this would be unsafe. Why would you want to probe anything smaller?

pediegoivan, 2020-04-26:
I assume that I am not the only one with this issue but, my printer has an inner area where the effector has a nearly perfectly planar movement (lets say R<100 mm) where I used to calibrate my printer with G33. I could get StDevs under 0.03 in that area.
Afterwards I used to run G29 on the whole bed that has a radius of 125 mm and compensated the non planar movements of that parts of the bed with UBL.
With the current firmware I get st devs higher than 0.12 if I calibrate my printer with a bigger radius, or I get a mesh considerably smaller than my bed.

sjasonsmith, 2020-04-26:
Isn’t that just hiding a problem? By calibration over as much of the bed as possible you ideally don’t even need to use any bed leveling. Is the issue that your bed is flat in the center but not toward the edges?

I’m not trying to devalue your suggestion or opinion here, just legitimately trying to understand the reasoning. I was involved in the change that removed the explicit radius. It happened at a time when delta probing was broken in many different ways, and it seemed redundant and made delta setup more complex. I’m sure it could come back again if there is real reason to do so.

pediegoivan, 2020-04-26:
I have a glass bed over a heated aluminum plate, I would bet that it is flat. The G29 mesh results visualized in the meshvisualizer plugin of octoprint show 3 high spots aligned with the towers and 3 low spots between the towers with R>100mm.
This comes from a compliant effector or a bad delta geometry, I know. But I have always compensated this with software... And doing this has become harder without having both parameters separately editable.

sjasonsmith, 2020-04-26:
@pediegoivan, can you add a workaround to your code to get the capability back, then report back whether the situation is improved for you?

In `Marlin/src/module/delta.cpp` change the `delta_calibration_radius` function to the following:

```
  float delta_calibration_radius() {
    return calibration_radius_factor * (
      #ifdef DELTA_CALIBRATION_RADIUS
        DELTA_CALIBRATION_RADIUS
      #elif HAS_BED_PROBE
        FLOOR((DELTA_PRINTABLE_RADIUS) - _MAX(HYPOT(probe.offset_xy.x, probe.offset_xy.y), MIN_PROBE_EDGE))
      #else
        DELTA_PRINTABLE_RADIUS
      #endif
    );
  }
```

You will also need to comment out the sanity check you will hit due to DELTA_CALIBRATION_RADIUS being present.

This should allow you to define DELTA_CALIBRATION_RADIUS in one of your configuration files. If that makes things better for you, we can discuss further whether the capability should restored. At the time @thinkyhead and I removed it there was very little input from any other delta users, so we just had to make some tough calls based on the information we had. We had assumed probing at the maximum possible radius would always be preferred for calibration purposes, and removing the option made delta configuration a little simpler.

pediegoivan, 2020-04-26:
OMG thank you! I will try this today and report back.

Thank you so much for doing this and so quickly! :)

sjasonsmith, 2020-04-26:
> Thank you so much for doing this and so quickly! :)

Well, I am sensitive to issues I made have inadvertently caused :)

When I started on the delta probing changes, non-zero probe offsets could easily crash the effector into towers. With the ability of M851 to dynamically change probe offsets it became important for this to become more dynamic.

I had presumed that printers with reduced calibration radiuses were doing so to prevent crashes due to offset probes, and it would no longer be important to define a reduced radius.

CharlieTemplar, 2020-04-26:
I run with reduced calibration radius only since moving to Marlin 2.0.x.  wider radius just doesn't seem to level correctly
I only have a small delta with 220mm bed, but with fittings to hold down a glass bed and the size of the heater block I can't reach further than 85 radius towards the towers.  
I was using DELTA_CALIBRATION_RADIUS set with M665 of 80mm in Marlin 1.1.9 and G33 gave me low stdDev (0.009).
Since moving to Marlin 2.0.x last year to work with a new SKR 1.3, G33 couldn't get better that 0.1 stdDev unless I reduced the DELTA_CALIBRATION_RADIUS to 60mm and even then it would print as though the bed wavy with high and low spots all over.
2.0.5.3 seems to have resolved the waviness and now G33 leaves me printing dead flat.  In fact I am currently 2hrs into printing PETG straight on glass with NO adhesive and no warping.  I only intended to print first layer as a test for how evenly it went down, and it was so good, I let it finish.
I'm yet to try calibrating with bigger calibration radius, but that's why I'm here... I would like to adjust the value without rebuilding the firmware.

CharlieTemplar, 2020-04-26:
My other issue is that not only is DELTA_CALIBRATION_RADIUS no longer a thing, it can not be adjusted but more importantly, the M503 command no longer shows the current value of M665 B
I have a plugin for octoprint that automates probing 10 points using G30 and the calculating M665 and M666 using least-squares routine in javascript.
This plugin uses M503 to get the current calibration parameters (M665&M666 output) but this no longer includes any reference to bed radius (printable or probeable)
I'm yet to test whether G30 still works in the same way, has anything changed for that command since last year?  it has changed quite a bit over the years/versions

pediegoivan, 2020-04-26:
@sjasonsmith I can confirm that the code that you gave me did the trick. I have just added it and disabled the sanity check as you pointed me out.
I look forward to have this merged back to Marlin as it is so useful to me (and I hope many more).
@CharlieTemplar I am sorry dude, but this code clearly does not add back the B parameter to the M665 gcode.

pediegoivan, 2020-04-26:
@CharlieTemplar I suppose that you could add the B report back to the GcodeSuite::M665 function at the M665.cpp file.
But I am not a very experienced coder and I am still not very familiarized with Marlin´s code.

sjasonsmith, 2020-04-27:
What if `DELTA_CALIBRATION_RADIUS` and `M665 B` came back, but the meaning changed to *maximum* delta calibration radius?

By default, this would be the entire printable radius.

At runtime, the current probe offsets (`M851`) and `MIN_PROBE_EDGE` would be used to determine the _actual_ calibration radius.

`M503 S0` would report back the configured radius (defaulting to printable radius), or the user-configured override. Reporting back the calculated radius is problematic due to it changing based on `M851`.

`M665 B` could then be used to decrease the maximum. Trying to increase it would not work, because that would exceed the safe probing area of the bed and you would get coerced back to the default value at runtime.

CharlieTemplar, 2020-04-27:
Sounds like it could work, but would it be possible to find out the value of MIN_PROBE_EDGE in gcode?

sjasonsmith, 2020-04-28:
@thinkyhead, what do you think of my proposal (see my most recent reply) to resolve issues related to the removal of a configurable `DELTA_CALIBRATION_RADIUS`? If you think this is a good idea I am willing to implement it.

@CharlieTemplar, this wouldn't be able to tell you MIN_PROBE_EDGE, but my proposal could be changed to default to "Printable radius - MIN_PROBE_EDGE".

Querying MIN_PROBE_EDGE, PRINTABLE_RADIUS and other non-configurable settings would be out of scope for this change. Those would be part of a broader solution to query printer information that goes far beyond delta configuration.

sjasonsmith, 2020-04-28:
@CharlieTemplar look at #17706 and #16741 for some insight and discussion regarding reporting printer information.

thinkyhead, 2020-05-06:
I propose that we add new options so that every constant in Marlin is a configurable variable with its own G-code and menu edit item. Make it so!

sjasonsmith, 2020-05-06:
> I propose that we add new options so that every constant in Marlin is a configurable variable with its own G-code and menu edit item. Make it so!

Lol, this case is unique because functionality went away with the changes we made a few months ago. At the time it didn't seem there were use-cases for it, but here we have two people claiming otherwise.

I didn't want to spend time implementing anything unless you thought it was a good idea @thinkyhead. It would ultimately be your call whether to let it in or not, so I'd like an opinion on whether it should be done or everyone just had to adapt to the change.

sjasonsmith, 2020-05-06:
We could also just add the constant back to the header, without the changes to M665. That would be pretty trivial.

dchauran, 2020-10-13:
NBot only did the functionality go away, but the same gcode was repurposed for something else, which is very troublesom. I would love to see the docs updated to reflect the current state. M665 BXXX now is used for rod correction, and this resulted in quite a lot of head scratching, because I thought I was setting the probing radius.

FWIW, as a delta user, I would love to be able to *temporarily* reduce my probing radius. For example, my probe is centered under the nozzle. I want to probe as far as possible, but until the initial calibration is dialed in this results in my probe running off the edge of the bed, so for initial probing I might use 125mm as opposed to the 130mm I would use for final calibration.

LVD-AC, 2021-08-28:
(1) When I first developed G33 I was printing on a fixed alu bed. I noticed that with some bad luck one or more probing points could fall on a small flaw in the bed and that by changing the calibration radius even so slightly I would get much better results. Hence I included it in M665 B to make it user selectable. Now that I'm printing on a removable mirror that is put on the fixed bed, I just give it a little twist if that happens. I'm more the set it and forget it type so I always calibrate on a radius 5mm smaller than the max printing radius to avoid running into the belts by the initial probing as stated above.

I'm all in favour to bring back a M665 B or call it P (for probe radius) to make the calibration radius user selectable so one can choose what works best in the given circumstances.

@sjasonsmith "What if DELTA_CALIBRATION_RADIUS and M665 B came back, but the meaning changed to maximum delta calibration radius?"

That could be a good way forward : a default value that is calculated from other parameters for DELTA_CALIBRATION_RADIUS that can be overridden by the user to a smaler value if wanted in a variable that is stored again in EEPROM.

(2) M665 originally was used to set parameters that need to be calibrated. I fail to understand why one included individual rod corrections but not individual delta radius corrections. If your rods have slightly different lenghts; so will the delta radii be slightly different. And if you correct for both the effect on calibration will be almost completelty canceled out, so why bother? Furthermore if your rods and delta radii are different the tower angles will change as well. On a delta everything is connected, that makes it so hard to calibrate them: you change one thing and 2 or 3 other things change with it.

Is is common practice to compensate for those small differences by changing the tower angles who's effects are not canceled out by the corresponding rod lenght and delta radius changes. Anyhow if you set indiviual rod lenghts G33 will make the necessary changes to the tower angle corrections to compensate for those, so you might as well set your rod lenght corrections to zero. IMO including these rod lenght corrections as user selectable in the M665 A B C parameters just adds to the confusion and do not add any fuctionality one can not achieve with the tower angle corrections.

(3) I no longer follow every change made; but if my insights are required on G33 just include me in a comment so I get an e-mail allert and I'll respond.

LVD-AC, 2021-09-18:
This issue can be closed, #22707 was merged

closed 2021-09-18 by thisiskeithb