MarlinFirmware/Marlin #17730: OctoPrint forces M112 if sees "Error:Probing Failed" on serial even if G29_RETRY_AND_RECOVER is enabled

o-nix, 2020-04-26:
STR:
* `G29_RETRY_AND_RECOVER` is enabled and `G29_MAX_RETRIES` set to 3, `G29_HALT_ON_FAILURE` is disabled
* OctoPrint is running and connected, no extra plugins for host commands or error handling
* Start a G29 routine
* When the probe is failed, then OctoPrint sends `M112` immediately

ER: G29 routine is restarted

AR: the printer is halted

FIX: lift probe error logic on the layer above by commenting or guarding the following line
https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/module/probe.cpp#L760

thinkyhead, 2020-04-27:
Is Octoprint configured to panic when it sees the word "Error"? If so, try disabling that option.

o-nix, 2020-04-27:
@thinkyhead I see that `SERIAL_ERROR_MSG` is used in `kill()` as well, so it seems it is the only "standard" way to understand, that the printer halted itself.

o-nix, 2020-05-02:
@thinkyhead BTW this is the OctoPrint's setting you mentioned, "Disconnect" and "Send M112" are set by default
[image]

boelle, 2020-06-20:
@o-nix still an issue?

o-nix, 2020-06-21:
@boelle uhm, of course. Wrong error reporting logic. I overrided it manually by commenting the line inmy local Marlin sources.

closed 2020-06-21 by thinkyhead

o-nix, 2020-06-21:
So, I confirmed that it is still an issue and you are closing it?
Would you prefer having a PR from me with the commented line?

boelle, 2020-06-21:
yes please, we welcome PR's