MarlinFirmware/Marlin #17731: [BUG] LPC1769 SKR 1.4 Turbo fails to compile

mochoandre, 2020-04-26:
I Am trying to compile latest 2.0.x bugfix for LPC1769 SKR 1.4 Turbo in vscode platformio but it fails to compile.

`> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\mocho\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1769; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.6.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy 
the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Marlin\src\HAL\LPC1768\HAL_SPI.cpp:216:6: error: no declaration matches 'void SPIClass::beginTransaction(SPISettings)'
  216 | void SPIClass::beginTransaction(SPISettings cfg) {
      |      ^~~~~~~~
In file included from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:52:
Marlin\src\HAL\LPC1768\include/SPI.h:42:10: note: candidate is: 'void SPIClass::beginTransaction(const SPISettings&)'
   42 |     void beginTransaction(const SPISettings&);
      |          ^~~~~~~~~~~~~~~~
Marlin\src\HAL\LPC1768\include/SPI.h:39:7: note: 'class SPIClass' defined here
   39 | class SPIClass {
      |       ^~~~~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
============================================================ [FAILED] Took 6.13 seconds ============================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
[... 159 lines cut ...]
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_e3              IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
malyan_M300               IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
======================================================= 1 failed, 0 succeeded in 00:00:05.069 =======================================================`

hamster65, 2020-04-26:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/17724

mochoandre, 2020-04-26:
Sorry did not see that

ellensp, 2020-04-27:
Please close this issue.

Talha909, 2020-04-27:
How you slove this issue

sjasonsmith, 2020-04-27:
@Talha909 look at the pull request linked above. Only a few small changes are needed to fix it. Hopefully it will be merged soon.

sjasonsmith, 2020-05-29:
@mochoandre, this issue could be fixed. Can you verify with bugfix-2.0.x and close this issue?

boelle, 2020-06-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-21 by boelle