MarlinFirmware/Marlin #17732: [BUG] UBL too low on far edge on X or Y axis outside of mesh

mobri2a, 2020-04-26:
### Bug Description

I am using UBL on 2.0.5.3 bugfix. Mesh inset is 30. I used UBL automatic leveling to do the initial probing, then edited each point using a 0.20mm feeler gauge. Prints are fine up to a few mm past the rightmost X or rearmost Y UBL points. Then I can see Z-axis motors rotate noticeably and drop the nozzle significantly. See attached photos - far right and rear edges. You can see the filament taper off.

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4535388/Configs.zip)

### Steps to Reproduce

1. Level the bed. Verify it with a feeler gauge
2. Print anything that goes past the inset by more than a few millimeters on X and Y axis

I'd suggest setting an inset to a high value. I'm using 30mm to avoid clips. Try 50. Generate an object to print that extends outside the inset. G26 will not show this issue.

**Expected behavior:** [What you expect to happen]

Print should be even

**Actual behavior:** [What actually happens]

Print "fades" into the bed. As the nozzle gets further outside the inset the Z-motors push the nozzle lower. There is no data in the mesh that could reasonably imply that the bed is lower outside of the mesh.

[image]

| | 0 | 1 | 2 | 3 | 4 | 5 | 6 
| --- | --- | --- | --- | --- | --- | --- | --- |
| **6** | 0.625 | 0.53 | 0.455 | 0.435 | 0.375 | 0.39 | 0.345 |
|**5** | 0.59 | 0.47 | 0.424 | 0.395 | 0.395 | 0.43 | 0.405 |
| **4** | 0.505 | 0.45 | 0.405 | 0.385 | 0.375 | 0.385 | 0.32 |
| **3** | 0.41 | 0.39 | 0.34 | 0.335 | 0.34 | 0.32 | 0.295 |
| **2** | 0.32 | 0.29 | 0.285 | 0.195 | 0.325 | 0.33 | 0.32 |
| **1** | 0.25 | 0.29 | 0.265 | 0.225 | 0.335 | 0.345 | 0.33 |
| **0** | 0.18 | 0.12 | 0.115 | 0.15 | 0.28 | 0.3 | 0.32 |

#### Additional Information

[image]
[image]
[image]

XDA-Bam, 2020-04-27:
After a quick look in the code, this does look like a design decision. The problem is: What do you assume the Z-correction should be outside of the mesh? Should the correction be extrapolated (keeping the last known slope)? Or should the correction be kept constant (last known value)? Or should we assume there's no correction at all outside the mesh?
Currently, the last assumption is made (see [here](https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/feature/bedlevel/ubl/ubl_motion.cpp#L60)). Additionally, the user can hard code the Z-offset outside the mesh using `UBL_Z_RAISE_WHEN_OFF_MESH`. That's probably not the best solution, but it looks to me like it works as designed.

Also, the problem is reinforced by the way you leveled the bed. That's not to say you're at fault (you're not), but let me explain: Your bed is somewhat tilted. Thereby, the correction grows to the corners of the mesh, especially on the back side. At the mesh boundary, it's close to half a millimeter and then - due to the current implementation - drops to zero directly outside of the meshed region. So as a first workaround, I'd recommend getting that tilt out of your bed and leveling it by hand until the outermost points of the mesh are closer to the average. Lower the rear left corner, raise the front left a bit and it should be close already. Then, you can additionally `#define UBL_Z_RAISE_WHEN_OFF_MESH 0.35`. Choose the actual value by using an average of the outermost mesh values (0.35 is just an example).

That being sad, it looks to me like it would make more sense to use the outermost mesh values by default when moving outside of the mesh. Is there a reason we're not doing that in the code?

boelle, 2020-06-20:
@mobri2a is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle