MarlinFirmware/Marlin #17735: [FR] MMU2S Spool Join

Minims, 2020-04-26:
### Description

MMU2S can be used as spool join, so you don't need to change spool. It load automatically the 2nd spool when the first one end.
 
### Feature Workflow

Taken from Prusa Documentation : 

9.2 SpoolJoin
When printing in Single mode with filament sensor turned on : 
1. the SpoolJoin function can be switched on in ​LCD Menu - Settings - SpoolJoin [on/off].​This function enables automatic switching of filament spools in case one of the filaments runs out. This is especially useful when you want to get rid of filament scraps that are still on spools without paying close attention to the printer during the printing process.
2. The SpoolJoin function cycles through all filament slots in a loop, so e.g. if you run out of filament in slot 1, the printer will automatically move to slot 2. In the meantime, you can replace the spool of filament in slot 1, so once the printer reaches slot 5, it will go back to slot 1 and continue printing.

#### Additional Information

https://prusa3d.com/downloads/manual/prusa3d_manual_mmu2_eng_1_02.pdf
Go To 9.2 Spool Join

Minims, 2020-04-26:
Hello @tonihoang, I saw your last PR about MMU2S sensor. Many Thanks for that. 
Maybe you have enough knowledge in Marlin and MMU2S to look at this feature if you have some time. Thanks ;-)

eoyilmaz, 2023-12-25:
I'm porting this feature as a part of #26523 👍

eoyilmaz, 2024-01-05:
This is now implemented as a part of the MMU3 feature efforts #26523 and it is working fine. PR will follow soon.