MarlinFirmware/Marlin #17737: TOOLCHANGE_ZRAISE set to 0 causes Z Offset to increase after each tool change

Protean-Man, 2020-04-26:
When TOOLCHANGE_ZRAISE in ConfigADV is set to 0 the Z Offset is increased for the first layer after each tool change. When printing the first layer after a tool change is too high above bed and then next layer is corrected. This occurs after each tool change - I found it is slightly lessened on T0.

Events: 
1 - Tool Changing Printer
2 - Printing an object that uses more than one tool (Each Layer is a different tool, not different tools on same layer - as far as I can tell).
3 - After first tool lays down filament the tool change is called and second tool is selected
4 - The second tool's first layer Z Offset prints way too high above bed and then on next layers the Z Offset returns to normal.

Fixed: only if TOOLCHANGE_ZRAISE is set to a non-zero number. Set mine to 5 and no more issue.

I found this was happening with Bilinear Bed Leveling in older bug reports and I tried it with Bilinear on/off and nothing changed - just in case someone brings that up.

**Expected behavior:** After tool change the second/third/etc tool Z offset does not adjust
**Actual behavior:** TOOLCHANGE_ZRAISE set to 0 = After tool change the second/third/etc tool Z offset is increased for first layer after each tool change.

studiodyne, 2020-04-28:
Please try the last bugfix , i had the same issue , and now it's fixed
[17229](https://github.com/MarlinFirmware/Marlin/issues/17229)

Protean-Man, 2020-04-30:
Update:
Maybe it was the wrong source originally or maybe there are two separate bugs but TOOLCHANGE_NO_RETURN being active also causes the exact same issue. When commented out the issue disappears.

studiodyne, 2020-04-30:
I will try on my machine , and i will back soon
Tell me the number of extruders , if fixed , if z_offset settings between them

@thinkyhead  , Help , so many times i heard here , some z offset issue , but , in tool change i see a computation about ' possible to raise or not ' , and , in cfg.h , you  allow z_offset but , you don't inform that the toolchange engine will not allow it .  
For ex : Nozzle1= 0 et NOzzle2= 1  (offset) on a SOLID FIXED DUAL EXTRUDER 
While printing : layer =0.3 , nozzle2 cannot print , because  nozzle1 EXPLODE the bed , and toolchange abort raising
In tool change , the computation will abort raising , and print is dead
The tool z_offset must be MECHANICAL or with separate z axis head 'dual x carriage '

JUST A THEORY

Protean-Man, 2020-04-30:
I currently have 4 extruders setup but I have found the issue with 2 on an earlier setup.
Extruders are separate tools with a servo locking tool changer.
Thanks!

Protean-Man, 2020-05-16:
Came back to this issue today and finding that the main problem seems to be TOOLCHANGE_NO_RETURN in Configuration_adv.h

When active TOOLCHANGE_ZRAISE seems to be multiplied each layer if it's non zero. And something else happens when it is zero. 
Also in Config.h HOTEND_OFFSET_Z does not do anything when TOOLCHANGE_NO_RETURN is active but works as expected when TOOLCHANGE_NO_RETURN is non-active.

Is there something in the TOOLCHANGE_NO_RETURN script that is causing these issues?

McUtty, 2020-05-17:
For me, it doesn't go down after a manual tool change. (T0 / T1)
 
When printing with both heads on one layer, the "Zxx" in the gcode is missing the first time. That's why he stays on top.
In the second round he gets a "Zxx" because of the new layer.
Then it's right.

If only one head prints per layer, then everything is ok. He always gets a "Zxx" for the new layer.

If "ZRAISE_ ..." = "0", then it works too.

studiodyne, 2020-05-17:
Hi .... 
For each user here , I need these informations please , i will try to solve this
But need Information in this order , and for every one here 

----------------------------------------------
2.0 or Last bugfix 2.0 release ?
SingleNozzle ?
Z offset between hotends if no single nozzle ?
Delta or cartesian ?
And the most important question UBL or NOT ?
And after these informations try this
Test with UBL enabled and disabled
Test with no PARK M217 v0 
Test with park

--------------------------------------------
I had the same issue and i have solved the issue in the last bugfix , it's not a forgotten layer , it's an ubl amount not respected
I want to help , but , to help I need to know what are the common repetitive informations
And after this , i will use my printer , with a mesurement tool , and step after step , i want to localise how to make the issue , and if it's a bug , i will have enough material to work

Thk you very much for the time you give

McUtty, 2020-05-17:
2.0 or Last bugfix 2.0 release ? - Marlin bugfix-2.0.x (download 08.05.2020)
SingleNozzle ? - dual x carriage with singlenozzels
Z offset between hotends if no single nozzle ? - Zoffset = 0
Delta or cartesian ? - cartesian 
And the most important question UBL or NOT ?  - No UBL

studiodyne, 2020-05-17:
@McUtty , it's ok , i'm waiting the others

Protean-Man, 2020-05-17:
2.0 or Last bugfix 2.0 release - **020005**
SingleNozzle - **Single Nozzle - 4 Tools - SWITCHING_TOOLHEAD**
Bltouch or switch ? For homing z - **Inductive Sensor**
Z offset between hotends if no single nozzle - **HOTEND_OFFSET_Z { 0.0, -1.5, -0.4, -1.0 }**
Delta or cartesian - **Cartesian**
And the most important question UBL or NOT **AUTO_BED_LEVELING_BILINEAR**

And after these informations try this
Test with UBL enabled and disabled - **Have tested with UBL and no change**
Test with no PARK M217 v0 - **No change, was tested when filament runout sensors were not active which required Advanced Pause/park**
Test with park - **Current situation and no change.**

I am finding that the Z height issues is active with TOOLCHANGE_NO_RETURN.

boelle, 2020-06-20:
@Protean-Man still an issue?

Protean-Man, 2020-06-21:
@boelle I will test with latest release and let you know in a day or two.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2020-07-04:
@Protean-Man (or others who have seen these issues), can you provide your configuration files that can be used to reproduce issues? Ideally configured to work with the latest `bugfix-2.0.x` branch. The simplest configuration which reproduces the issue would be best (for example, 2 tools instead of 4).

It seems that perhaps some of these issues have already been fixed as well? An updates "Steps to Reproduce" would be very helpful, to identify an exact procedure to reproduce any remaining issues.

Protean-Man, 2020-08-05:
Issue has disappeared with recent BUG releases.
It was fixed a month ago but I did not update this GIT Issue - apologies.

closed 2020-08-05 by Protean-Man