MarlinFirmware/Marlin #17738: Dual extruder PID values not recognised in firmware[BUG] (short description)

seitmand, 2020-04-26:
### Bug Description

<I have two different PID values for my two extruders.  They are defined in firmware, but do not load.  Only the second set of values is stored, and the PID values for extruder E0 is ignored.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce
1. Marlin firmware is uploaded to my printer (Artifex2Duo All Metal)
2. The M301 values for both extruders are examined.M503 values shown below.  [They are incorrect]
3.  M502 command is given to reload firmware values.
4.  M301 values again examined.  [Still incorrect]

I expect the values to reflect the E0 and E1 PID values as listed in my Configuration.h file.

Instead, only the PID values for E1 are stored.

#### Additional Information
`
echo: Last Updated: 2020-04-26 | Author: D.Seitman 3DMakerWorld
echo:Compiled: Apr 26 2020
echo: Free Memory: 2888  PlannerBufferBytes: 1504
echo:V77 stored settings retrieved (706 bytes; crc 22050)
echo:Active Extruder: 0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Active Extruder: 0
X:2.00 Y:2.00 Z:2.00 E:0.00 Count X:160 Y:160 Z:3024
ok
echo:Active Extruder: 0
echo:Active Extruder: 0
X:2.00 Y:2.00 Z:2.00 E:0.00 Count X:160 Y:160 Z:3024
ok
echo:endstops hit:  Z:1.98
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.21 Y80.21 Z1511.90 E473.20
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E150.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X6000.00 Y6000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Hotend offsets:
echo:  M218 T1 X35.50 Y1.74 Z0.000
echo:; Material heatup parameters:
echo:  M145 S0 H185 B70 F0
echo:  M145 S1 H230 B100 F0
echo:; PID settings:
echo:  M301 E0 P21.40 I1.57 D72.75
echo:  M301 E1 P21.40 I1.57 D72.75
echo:  M304 P372.36 I30.10 D1151.54
echo:; Tool-changing:
echo: Z2.00
ok
echo:Hardcoded Default Settings Loaded
ok
echo: e:0 p:21.40 i:1.57 d:72.75
ok
echo: e:1 p:21.40 i:1.57 d:72.75
ok
[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/4536688/Configuration_files.zip)
`

seitmand, 2020-04-26:
If I understand the logistics of github, I have used the commit #17708 and all the files included up to that branch.

Roxy-3D, 2020-04-26:
Personally...   I think we should have multiple values for the different nozzles on a machine.  I think we should change the section:
````
  // Ultimaker
  #define DEFAULT_Kp 22.2
  #define DEFAULT_Ki 1.08
  #define DEFAULT_Kd 114
```
to have information for all nozzles that a machine has.    Most multi-head machines have identical extruders.   So one set of numbers works for those machines.   But I know my 2 tool machine has nozzles that look identical but have significantly different PID numbers.

Your thoughts?

seitmand, 2020-04-27:
I think it is vital.  There is supposed to be a mechanism, but it doesn't work.  When I manually set the values and then store them to EEPROM, they remain correct.  Only the values listed in the firmware are not ported properly.

Originally, my extruders were close enough to allow one set of PID numbers for both (I think the precision of these numbers way exceeds the accuracy needed by the machine).  However, I recently switched one of my heaters from a 30W to a 40W, so my PID numbers are much different.

boelle, 2020-06-20:
@seitmand is the issue still there?

seitmand, 2020-06-21:
It was in the last version i downloaded. Give me a day to download latest
version to see if the problem was corrected.

Thanks
David

boelle, 2020-06-23:
This is the place [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) its updated daily

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

thinkyhead, 2020-06-30:
Sorry, this should have been followed up with assistance to get closer to the bottom, instead of closed.

- So you have enabled `PID_PARAMS_PER_HOTEND` ?
- And, if you do `M301 E1 P<number>` you get back the same value when you do another `M301 E1`?

I've only had a casual look at the code, but so far it looks okay.

sjasonsmith, 2020-07-04:
@thinkyhead I noticed you added `DEFAULT_Kp_LIST` a couple days ago, do you expect that to be ready for @seitmand to try?

seitmand, 2020-07-04:
I think the loop which is not setting the PID values for each extruder needs to be fixed before looking at @sjasonsmith 's suggestion. If I was a better OOP programmmer, I'd attempt it myself, but I'm too old-schooled.

thinkyhead, 2020-07-15:
The loop to initialize the array looks healthy to me. I'll run debugging on it later.

seitmand, 2020-08-15:
@sjasonsmith - thank you for removing the stale label.

@thinkyhead - 
>And, if you do M301 E1 P<number> you get back the same P value when you do another M301 E1?
 
I may have missed this query but I hope you realized that this is one of the ways I determined that the PID values are not kept for each individual extruder.

AFAIK, this issue has not been fixed, but I have not downloaded the latest commit in a few weeks.  If one of the programmers believes this has been resolved, please let me know, and I will do the download.

Because I am using an apparently non-standard printer, I need to make a number of modifications to the files (mostly dealing with the temperature sensors), so the process is a bit more complicated for me than to simply copy my configuration files.  Hence, I don't download new commits often.

seitmand, 2020-08-30:
I am still waiting to hear that appropriate changes have been made to Marlin to remove this bug.

Thank you.

sjasonsmith, 2020-08-30:
@seitmand, try replacing your defaults like this:

Old way:
```
  #define DEFAULT_Kp 21.73
  #define DEFAULT_Ki 1.54
  #define DEFAULT_Kd 76.55
```

New way:
```
  #define DEFAULT_Kp_LIST {21.73, 10.0}
  #define DEFAULT_Ki_LIST {1.54, 0.0}
  #define DEFAULT_Kd_LIST {76.55, 100.2}
```

Of course the number of elements in each list should match your number of hotends.
You also need to be sure to enable `PID_PARAMS_PER_HOTEND`.

Please report back how that goes. It currently isn't documented in the configuration file, so if it works there is some improvements needed in that area.

sjasonsmith, 2020-08-30:
Oh, and you should try that on 2.0.6.1. This is a fairly recent addition, so if you are running old firmware it isn't going to work. It seems to have gone in about 2 months ago.

seitmand, 2020-08-30:
@sjasonsmith The "new way" is how I have entered my PID values.  But, I have not tried 2.0.6.1 yet.  Give me a day or two to upload and revise my files.  If the problem persists, I'll attach a copy of my latest configuration files; if the problem was solved, I'll note that here also.

sjasonsmith, 2020-08-30:
Remember to re-initialize your EEPROM to try out the new feature. You will already have old values stored there.

seitmand, 2020-08-30:
I was planning to store fake values for E0 and E1 before loading the new firmware.  This way I'd be able to tell if the M502 / M500 commands actually wrote the correct values in.  Is there another way to re-intiialize the EEPROM?

Thanks

sjasonsmith, 2020-08-31:
That should work fine. You can also do it through LCD menus.

seitmand, 2020-09-02:
Just to let you know I am still working on loading the latest firmware.  I've run into a compilation error `fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino Mega or Mega 2560.
`
I am trying to get the  number of characters in the compile command down below the (approx) 32K which I think is causing this error.

seitmand, 2020-09-04:
Actually, I'm having some problems, and, yes, I am building using Arduino IDE.  To better see what is happening, I have changed my settings to verbose on - More.

The filename extension error message mentioned above was received when I mistakenly put parentheses around the PID values, instead of braces.

Now that I have braces, my statements look like:
>     #define  DEFAULT_Kp  {10.0,  22.0}
>     #define  DEFAULT_Ki  {0.5,  1.60}
>     #define  DEFAULT_Kd {54.7, 73.0}

I am receiving the following error messages:

> In file included from c:\myarduino\sketch\src\core\boards.h:24:0,
>                  from c:\myarduino\sketch\src\inc\MarlinConfigPre.h:35,
>                  from c:\myarduino\sketch\src\inc\marlinconfig.h:28,
>                  from C:\MyArduino\sketch\src\module\settings.h:28,
>                  from C:\MyArduino\sketch\src\module\settings.cpp:47:
> C:\MyArduino\sketch\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
> c:\myarduino\sketch\src\core\macros.h:247:40: error: braces around scalar initializer for type 'const float'
>  #define ARRAY_N(N,V...) { _LIST_N(N,V) }
>                                         ^
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
>  #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
>                                 ^~~~~~~
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
>  #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
>                                ^~~~~~~~~~~~~~~~
> C:\MyArduino\sketch\src\module\settings.cpp:2664:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
>            ARRAY_BY_HOTENDS1(DEFAULT_Kp)
>            ^~~~~~~~~~~~~~~~~
> c:\myarduino\sketch\src\core\macros.h:247:40: error: braces around scalar initializer for type 'const float'
>  #define ARRAY_N(N,V...) { _LIST_N(N,V) }
>                                         ^
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
>  #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
>                                 ^~~~~~~
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
>  #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
>                                ^~~~~~~~~~~~~~~~
> C:\MyArduino\sketch\src\module\settings.cpp:2670:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
>            ARRAY_BY_HOTENDS1(DEFAULT_Ki)
>            ^~~~~~~~~~~~~~~~~
> c:\myarduino\sketch\src\core\macros.h:247:40: error: braces around scalar initializer for type 'const float'
>  #define ARRAY_N(N,V...) { _LIST_N(N,V) }
>                                         ^
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
>  #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
>                                 ^~~~~~~
> c:\myarduino\sketch\src\inc\Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
>  #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
>                                ^~~~~~~~~~~~~~~~
> C:\MyArduino\sketch\src\module\settings.cpp:2676:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
>            ARRAY_BY_HOTENDS1(DEFAULT_Kd)
>            ^~~~~~~~~~~~~~~~~
> 
Any suggestions on how to continue?  I have never used Visual Studio Code with PlatformIO, so I'm sure that switching over will be a learning process.  I'm hoping that I can make this work with the Arduino IDE, which I have a long vested interest in.

Thanks

boelle, 2020-09-15:
https://www.youtube.com/watch?v=bPdZvv8jJJ0

marlin 2.0 is both 8 bit and 32 bit so this video is still usefull

seitmand, 2020-09-16:
Thank you for this video.  However, I'm a bit confused.  I don't use the
ramps board or the due.

seitmand, 2020-09-16:
Well, I believe I have installed Visual Studio and PlatformIO.  I must admit that I feel like I am working blind because I do not know all the options that are available.  However, using the files I as I already had them prepared, I get the following error when I try to compile:

> In file included from Marlin\src\module\../inc/../core/boards.h:24:0,
                 from Marlin\src\module\../inc/MarlinConfigPre.h:35,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:47:
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\../inc/../core/macros.h:247:40: error: braces around scalar initializer for type 'const float'
 #define ARRAY_N(N,V...) { _LIST_N(N,V) }
                                        ^
Marlin\src\module\../inc/Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
 #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
Compiling .pio\build\mega2560\src\src\sd\SdFatUtil.cpp.o
                                ^
Marlin\src\module\../inc/Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
 #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
                               ^
Marlin\src\module\settings.cpp:2664:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
           ARRAY_BY_HOTENDS1(DEFAULT_Kp)
           ^
Marlin\src\module\../inc/../core/macros.h:247:40: error: braces around scalar initializer for type 'const float'
 #define ARRAY_N(N,V...) { _LIST_N(N,V) }
                                        ^
Marlin\src\module\../inc/Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
 #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
                                ^
Marlin\src\module\../inc/Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
 #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
                               ^
Marlin\src\module\settings.cpp:2670:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
           ARRAY_BY_HOTENDS1(DEFAULT_Ki)
           ^
Marlin\src\module\../inc/../core/macros.h:247:40: error: braces around scalar initializer for type 'const float'
 #define ARRAY_N(N,V...) { _LIST_N(N,V) }
                                        ^
Marlin\src\module\../inc/Conditionals_LCD.h:554:32: note: in expansion of macro 'ARRAY_N'
 #define ARRAY_BY_HOTENDS(V...) ARRAY_N(HOTENDS, V)
                                ^
Marlin\src\module\../inc/Conditionals_LCD.h:555:31: note: in expansion of macro 'ARRAY_BY_HOTENDS'
 #define ARRAY_BY_HOTENDS1(v1) ARRAY_BY_HOTENDS(v1, v1, v1, v1, v1, v1, v1, v1)
                               ^
Marlin\src\module\settings.cpp:2676:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
           ARRAY_BY_HOTENDS1(DEFAULT_Kd)
           ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\module\settings.cpp.o] Error 1
========================================================== [FAILED] Took 43.68 seconds ==========================================================

>Environment    Status    Duration
-------------  --------  ------------
>mega2560       FAILED    00:00:43.685
===================================================== 1 failed, 0 succeeded in 00:00:43.685 ===================================================== 
>The terminal process "C:\Users\seitm\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

ellensp, 2020-09-16:
I have no idea what you have done. But here are your original Config files updated for current bugfix. It compiles fine on platformio
[Updated Configuration for bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5229790/Updated.Configuration.for.bugfix.zip)

ellensp, 2020-09-16:
you cannot put DEFAULT_Kp  values in braces...

ellensp, 2020-09-16:
you must use

    #define  DEFAULT_Kp_LIST  { 8.73,  21.4 }  //  8.61, 25.52  //22.61 //15.0
    #define  DEFAULT_Ki_LIST  { 0.35,  1.57 } //  0.33,  2.04  // 1.69 // 2.5
    #define  DEFAULT_Kd_LIST { 54.91, 72.75 }  // 55.35, 79.95  //75.95 //60.0    

NOTE: the _LIST suffix

ellensp, 2020-09-16:
M503 output
echo:  M301 E0 P8.73 I0.35 D54.91
echo:  M301 E1 P21.40 I1.57 D72.75

seitmand, 2020-09-16:
@ellensp - My sincere apologies.  I totally missed the "_LIST" change to the PID value lines.  That totally explains why I was getting error messages for the DEFAULT values.

I will not be able to test this on my printer until next week, but I will report back if the problem has been corrected and solved.  Thank you

boelle, 2020-09-16:
> I don't use the ramps board or the due.

but istalling the software will be the same

ellensp, 2020-09-17:
multiple PIDs example is now part of Example Configuration. Closing.

closed 2020-09-17 by ellensp