MarlinFirmware/Marlin #17739: PID_FAN_SCALING no effect

jimhiggs, 2020-04-27:
My parts fans seriously cool my nozzle. Previously I ran PID Auto Tune with the fans running. Having just updated to the latest bugfix 2 my nozzle was cooling despite having the same autopid values in configuration.h. While investigating I noticed the addition of PID_FAN_SCALING to configuration.adv so I thought I should try that. I enabled it and also enabled ALTERNATIVE DEFINITION.  
I noticed that M301 now includes an F parameter not specified in the Marlin Gcode docs but only when PID_FAN_SCALING is enabled so I assume this is to set the DEFAULT_Kf ? Is that correct? Only I Pid autotuned with that set to 0 and  then heated nozzle to 200 and tried different F values while turning on the Fan 100%.
Did not have any effect on falling nozzle temp, I tried up to F60 .

mathis92, 2020-04-30:
I am also not able to get this to work. Anybody else has made it to work ?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle