MarlinFirmware/Marlin #17742: [Bug] firmware ignoring dual extruder offset im change

mvthedemon, 2020-04-27:
Offsets for dual extruder are in firmware yet in tool change all I get is a z raise if 2mm and I've tried a number of things to fix 

On tool change from tool0 to tool1 shift -35 on x axis to line up second extruder and continue 

Raises 2mm in z axis and never shifts in the x 
SKR PRO V1.1 marlin 2.0.5.3 and 2.0 bugfix 

Worked previously on 1.1.7/8

thinkyhead, 2020-04-27:
Please fill out the template. We can't follow up on issues with no information.

Geeks who like to mess with tool-changing include @InsanityAutomation and @studiodyne. I am afraid to touch it because I don't have any fancy machines.

studiodyne, 2020-04-27:
Please can you try ,  Marlin 2.0 release first , and verify if it works and try after the bugfix2.0  , because , very important for me to have this information. Some minors reviews have been made in recent bugfix , but not enough to make your issue. May be an old bug .
Thks

mvthedemon, 2020-04-27:
I will upload config  later this afternoon at work currently but side note it has been a persistent problem from 2.0 release and bugfix

mvthedemon, 2020-04-28:
Ok here is my config.h and config-adv.h #

mvthedemon, 2020-04-28:
Was a bad sd card insert facepalm here thanks guys

closed 2020-04-28 by mvthedemon