MarlinFirmware/Marlin #17743: Power loss recovery issue[BUG] (short description)

github051, 2020-04-27:
Currently, i'm using the marlin 2.0.3 version (Controller: MKS Robin) with power loss recovery and is working fine but when the power comes if i click on resume the hot end is moving home without heating the nozzle. why ? any idea ?

Requirement: if i click resume then the nozzle heats up first then go to home

chepo92, 2020-04-30:
I think you should:
1. Use the template for issues
2. Test with bugfix 2.0 branch, 2.0.3 is a bit outdated

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle