MarlinFirmware/Marlin #17745: MKS robin2 Wont compile Platform IO Bugfix 2.0.x

wookie666, 2020-04-27:
trying to compile marlin bugfix latest for a MKS Robin2 using VS Studio and platform io. 
I get the following compile errors.

In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp:28:
Marlin\src\HAL\STM32_F4_F7\../../inc/../HAL/./STM32_F4_F7/HAL.h:51:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32_F4_F7\HAL.cpp:26:0:
Marlin\src\HAL\STM32_F4_F7\HAL.h:51:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp:28:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:31:0: warning: "BOARD_NAME" redefined
   #define BOARD_NAME "MKS_ROBIN2"

<command-line>:0:0: note: this is the location of the previous definition
In file included from Marlin\src\HAL\STM32_F4_F7\STM32F4\timers.cpp:25:0:
Marlin\src\HAL\STM32_F4_F7\STM32F4\../HAL.h:51:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32_F4_F7\../../inc/../HAL/./STM32_F4_F7/HAL.h:51:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:31:0: warning: "BOARD_NAME" redefined
   #define BOARD_NAME "MKS_ROBIN2"

<command-line>:0:0: note: this is the location of the previous definition
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp:28:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: error: 'BOARD_INFO_NAME' was not declared in this scope
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\STM32F4\timers.cpp.o] Error 1
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: note: suggested alternative: 'BOARD_NAME'
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: error: 'BOARD_INFO_NAME' was not declared in this scope
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp.o] Error 1
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: note: suggested alternative: 'BOARD_NAME'
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
         ^~~~~~~~~~~
         transfer
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:157:9: error: 'class SPIClass' has no member named 'dmaSend'
     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
         ^~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\HAL_SPI.cpp.o] Error 1

ellensp, 2020-04-27:
SERIAL_PORT cannot be 0. (Port 0 does not exist.) 
Set SERIAL_PORT to something else

ellensp, 2020-04-27:
The internet says it should be "#define SERIAL_PORT 3" for your board.

wookie666, 2020-04-27:
sorry forgot to say. already went past that error. I still get.

`In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp:28:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:31:0: warning: "BOARD_NAME" redefined
   #define BOARD_NAME "MKS_ROBIN2"

<command-line>:0:0: note: this is the location of the previous definition
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:31:0: warning: "BOARD_NAME" redefined
   #define BOARD_NAME "MKS_ROBIN2"

<command-line>:0:0: note: this is the location of the previous definition
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp:28:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: error: 'BOARD_INFO_NAME' was not declared in this scope
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: note: suggested alternative: 'BOARD_NAME'
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\EmulatedEeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/pins.h:564:0,
                 from Marlin\src\HAL\STM32_F4_F7\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: error: 'BOARD_INFO_NAME' was not declared in this scope
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../pins/stm32f4/pins_MKS_ROBIN2.h:35:32: note: suggested alternative: 'BOARD_NAME'
   #define DEFAULT_MACHINE_NAME BOARD_INFO_NAME
                                ^
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/language.h:86:26: note: in expansion of macro 'DEFAULT_MACHINE_NAME'
     #define MACHINE_NAME DEFAULT_MACHINE_NAME
                          ^~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\../../inc/../core/../lcd/language/language_en.h:42:62: note: in expansion of macro 'MACHINE_NAME'
   PROGMEM Language_Str WELCOME_MSG                         = MACHINE_NAME _UxGT(" Ready.");
                                                              ^~~~~~~~~~~~
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
         ^~~~~~~~~~~
         transfer
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:157:9: error: 'class SPIClass' has no member named 'dmaSend'
     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
         ^~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\HAL_SPI.cpp.o] Error 1`

ellensp, 2020-04-27:
bug in Marlin/src/pins/stm32f4/pins_MKS_ROBIN2.h
change the following
```
#ifndef BOARD_INFO_NAME
  #define BOARD_NAME "MKS_ROBIN2"
#endif
```
into this
```
#ifndef BOARD_INFO_NAME
  #define BOARD_INFO_NAME "MKS_ROBIN2"
#endif
```

wookie666, 2020-04-27:
fixed that part. 
Now this 
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
         ^~~~~~~~~~~
         transfer
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\STM32_F4_F7\HAL_SPI.cpp:157:9: error: 'class SPIClass' has no member named 'dmaSend'
     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
         ^~~~~~~
*** [.pio\build\STM32F4\src\src\HAL\STM32_F4_F7\HAL_SPI.cpp.o] Error 1

ellensp, 2020-04-27:
please attach Configuration files and platformio.ini file.

wookie666, 2020-04-27:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4539480/Configuration.zip)
[pio.zip](https://github.com/MarlinFirmware/Marlin/files/4539481/pio.zip)

not sure exactly whats needed with the platform io file. ive atached everything i think is needed.

ellensp, 2020-04-27:
the real marlin doesnt seem to have support for this board. are compiling MKS code?
ie https://github.com/makerbase-mks/MKS-Robin2/tree/master/firmware/Marlin2.0-MKS-Robin2

wookie666, 2020-04-27:
Tried that too. It asks for me to install git. I wanted to try pure Marlin
to test because I'm getting knocking sounds on the MKS code with lv8729
drivers. Similar to this video https://youtu.be/VuU9CssOD0Y

I'm not sure what it is

sjasonsmith, 2020-04-27:
You probably shouldn’t be compiling source in the HAL/STM32_F4_F7 folder, which is where your latest errors are from. You should try using HAL/STM32 instead. This is true even if you have an F4 or F7 MCU.

wookie666, 2020-04-27:
I'll try that tomorrow. Any idea what the clicking /knocking noise would be
in that YouTube video I've linked. It's not as bad when running slower

wookie666, 2020-04-27:
im clicking eng:stm32f4 and then build. I cannot find a env:stm32 folder on its own

sjasonsmith, 2020-04-27:
Oh, that environment is just the for test I think. I doubt any board would actually work with it. If you want to use the Marlin repo with this board I suggest you go to the MKS repo to see what changes they made. You’ll have to change quite a few things for a new STM32 board.

wookie666, 2020-04-27:
yeah i would have to change heaps. Im just trying to figure out why i cannot run this at 1/128 steps without it sounding like its going into quad step mode. 
It sounds like its skipping steps. But its not losing position. 
Any idea?
Best i can do is 1/32 steps at 60mm second and even then. I still get the odd sound every second or so

sjasonsmith, 2020-04-27:
Running at 128 micro steps overloads even fast processors, unless you are printing at very slow speeds. Most of us seem to just run in 16 then let the TMC drivers interpolate to 256.

Micro stepping isn’t at all what this issue was opened for though, so I was responding to the build questions. Perhaps close this issue and go to the record channel or one of the other support venues for general help.

wookie666, 2020-04-27:
thanks mate. i think ill have to invest in TMC drivers.
Im currently using LM7806 MKS drivers. so ill just stick to 1/32 steps for
now.
and i have multiple issues. Im not going to open another case for an issue
ive already asked about. if you cant help me, ill find it elsewhere

github051, 2020-04-27:
> thanks mate. i think ill have to invest in TMC drivers. Im currently using LM7806 MKS drivers. so ill just stick to 1/32 steps for now. and i have multiple issues. Im not going to open another case for an issue ive already asked about. if you cant help me, ill find it elsewhere
> […](#)

With TMC 2208 I tried. But same issue. 6 months back I posted this issue to MKS but no response till now.Mks robin2 is worst board.

wookie666, 2020-04-30:
> > thanks mate. i think ill have to invest in TMC drivers. Im currently using LM7806 MKS drivers. so ill just stick to 1/32 steps for now. and i have multiple issues. Im not going to open another case for an issue ive already asked about. if you cant help me, ill find it elsewhere
> > […](#)
> 
> With TMC 2208 I tried. But same issue. 6 months back I posted this issue to MKS but no response till now.Mks robin2 is worst board.

Yeah im starting to think the same thing. it works on their firmware. But it clicks/knocks and has really weird sounds. Almost like acceleration is way too high. not sure. only happens on their board.
Also, octoprint will not print to it. it keeps freezing/pausing during printing. multiple times on one layer. Almost like it cant keep up or the board cant keep up.

boelle, 2020-06-20:
@wookie666 is this still an issue?

wookie666, 2020-06-20:
Not sure mate I gave up

boelle, 2020-06-20:
will close then, we can always reopen

closed 2020-06-20 by boelle

mbfab, 2020-07-05:
I have a similar problem, but i have a SKR Mini E3 v2.0

I have tried following to get a firmware that works, I have downloaded the firmware from the Bigtreetech, DONT Work,
My latest attempt I have downloaded the latest bugfix for Marlin 2.0.x, I also downloaded the latest Configurations for my printer and pasted in the 4 files into the marlin folder and clicked replace. I changed to the correct Motherboard in Configuration.h, and changed to the right card in platformio.ini... still i get this problems.

I get the following error codes.

In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/Servo.cpp:25:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
Compiling .pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_wired.cpp.o
In file included from Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/dogm/../../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp:29:
Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/HAL.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/dogm/../../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_swspi.cpp:21:
Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/eeprom_flash.cpp:32:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/HAL_SPI.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/eeprom_sdcard.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/dogm/u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_flash.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_sdcard.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/eeprom_wired.cpp:22:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:81:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/eeprom_wired.cpp.o] Error 1

Its really seems to be a mess to update the firmware, I have tried to resolve this for 2 days now, when i fix one fault another appears.

thisiskeithb, 2020-07-05:
@mbfab: There's no need to comment on random PRs and old unrelated issues.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).