MarlinFirmware/Marlin #17746: [BUG] Bed Temperatur tries to match hotend temp

jonaskonig, 2020-04-27:
### Bug Description
Bed Temperatur tries to match hotend temp

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[BUG.zip](https://github.com/MarlinFirmware/Marlin/files/4539479/BUG.zip)

### Steps to Reproduce

Works not every time but sometimes
1. edit your UBL-Mesch manually
2. See the temperature of the Bed rise and rise
3. Weit for runawayprotection to kick in 
4. That will never happen

**Expected behavior:** [What you expect to happen]
The Temperatur of the bed schould not match the Temperatur of the Hotend

**Actual behavior:** [What actually happens]
The temperature of the Bed rises and tries to Match the temp of the Hotend even the Tempsensor of the Bed is ignored for Heating. It seems like the Temp sensor is completely ignored in the firmware. Maybe it has to do with PID bedcontroll
[BUG.zip](https://github.com/MarlinFirmware/Marlin/files/4539479/BUG.zip)

#### Additional Information
So it happened to me two times know I'm not able to recreate the issue all the time, but it seems like something gets mixt up really bad and I don't want to know what's gonna happen if the Temp starts rising even more. Thermal runaway protection is not kicking in either so the firmware really thinks it's okay to heat up that much. What is confusing me is that there is a Max temp for the Bed but that is ignored to.

boelle, 2020-06-20:
@jonaskonig is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle