MarlinFirmware/Marlin #17748: [BUG] No compilation 'XY_FREQUENCY'  -->refresh_frequency_limit not declared in scope

studiodyne, 2020-04-27:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-27:
Please provide configuration files.

studiodyne, 2020-04-27:
hi , thanks to answer , i'm the author of the frequency limiter , but @thinkyhead have reviewed the code and , a missing declaration somewhere. Easy for him .... less for me ...
So I'm waiting
Thk you very much

closed 2020-04-27 by studiodyne