MarlinFirmware/Marlin #17756: Mesh validation G26 unsafe z end

tablatronix, 2020-04-27:
2.0 bugfix ender 3 default config

Set manual mesh leveling
G26

After finishing Z height does not clear safe distance, gouges bed.

[image]

tablatronix, 2020-04-27:
Code looks like it should be doing this ok, not sure whats going on here, but it definitely did not.

`#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points`

```C++
  retract_filament(destination);
  destination.z = Z_CLEARANCE_BETWEEN_PROBES;
  move_to(destination, 0);                                    // Raise the nozzle

  destination = g26_xy_pos;                                   // Move back to the starting XY position
  move_to(destination, 0);                                    // Move back to the starting position
```

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle