MarlinFirmware/Marlin #17757: [BUG] Sensorless homing - endstops always TRIGGERED

b3nnb, 2020-04-27:
### Bug Description

Sensorless homing failes to home. the Endstops always report as **TRIGGERED**.
When i try to home, i can hear the motors engage but not do anything. on the screen of the printer i see the Z distance jump from 250 to 260 for a split second.
I cannot inver the logic as sensorless homing requires the endstops to be **false**.
The axis will move up and down when controlling manually, all 3 towers move correctly.
I have tried a bunch of fixes that others have gotten to work but mine has not.
The sensitivity has been adjusted and test from 0 to 255 in increments of 20

This works when using mechanical endstops.

### My Configurations

Kossel mini
Bigtreetech skr v1.3
Bigtreetech tmc2209 v1.2
Marlin Bugfix 2.0.x

I have no jumpers under the drivers, i only have them on the red UART pins beside the steppers and on the pins beside the endstop connectors. all for X/Y/Z

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I will post the configs soon I have to get them off my other computer

### Steps to Reproduce

1. G28
2. Watch nothing happen

**Expected behavior:** [What you expect to happen]

All axis should home to the top and stop. setting that to the max z height

**Actual behavior:** [What actually happens]
homing does not occur, motors engage but done move

JohnnyTheOne, 2020-04-27:
Sensitivity TMC2209 Use M914 X Y Z to set the stall threshold at runtime
   *    HIGHEST       255  
   *    LOWEST         0     

CURRENT_HOME X_CURRENT to CURRENT_HOME  400

rglory, 2020-04-27:
I believe you need to flip logic of endstops. I have working sensorless  with SKR 1.3 and TMC2209 and normally (always) M119 shows X and Y endstops status as "open". Note TMC2209 keep signal low and raise up for short period of time when stallguard detected so you should never see them in TRIGGERED state.

b3nnb, 2020-04-27:
> CURRENT_HOME

> Sensitivity TMC2209 Use M914 X Y Z to set the stall threshold at runtime
> 
> * HIGHEST       255
> * LOWEST         0
> 
> CURRENT_HOME X_CURRENT to CURRENT_HOME 400

I tried the M914 as well as setting in the firmware, a bunch of settings between 0.and 255.
my current home is set to 700-800. thats usually where i set it for the steppers i have. do you think thats too high?

b3nnb, 2020-04-27:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4542489/Configuration.zip)

heres my config and config_adv

b3nnb, 2020-04-27:
> ~I believe you need to flip logic of endstops.~ I have working sensorless configuration with SKR 1.3 and TMC2209 and normally (always) M119 shows X and Y endstops status as **open**. Note TMC2209 keep DIAG signal low and raise up for short period of time when stallguard detected so you should never see them in **TRIGGERED** state.
> 
> UPDATE: I missed your message that you have X_MIN_ENDSTOP_INVERTING false. I do have it also false and my endstop statuses for X and Y (where I use 2209 and sensroless homing) are always open. Looks like something wrong with you hardware configuration. Most probably that DIAG pin is not connected to endstop. You probably need to trace it - it should be in low state (low voltage) normally. Just mesure voltage on those endstop pins that enables sensorless homeing for start.
> 
> Also with SKR 1.3 I had issue with motor shakes X or Y and then controller goes to halt state. It started to work properly after I turned TMC_DEBUG on. Hope that helps.

TMC_DEBUG is enbaled, when i run an M122 i get everything i would expect as an output like stallgaurd enabled and thr set to _whatever i set it to at thet time_.
what should i be testing and lookign for on the diag pin. i thought with combining the 2 bigtreetech pieces(driver and skr 1.3) that it would have been setup for that.

forgive me im not the most knowledgeable in the marlin world

rglory, 2020-04-28:
I am not sure than TMC_DEBUG is the only thing I changes though. But the fact is it had that behavior - no matter what I do, stepper just moved it left and right aliitle  and then controller hangs with bad state. I tried different things and then I changed TMC_DEBUG and probably something else, rebooted and suddenly it all started to work flawlessly. I used exactly the same configuration as you do - Bigtreetech skr v1.3 and Bigtreetech tmc2209 v1.2

Try to measure voltage on those DIAG pins, they should be low (mine has aroun 0.2V) and enstop status should be "open". If they are not low and connection is fine then looks like you need to deal with some settings to make that pin on TMC driver active.

rglory, 2020-04-28:
Wait a sec, I am wrong. I have SKR v1.4 from Bigtreetech, but 1.3 is from Witbot:
https://www.amazon.com/gp/product/B07P72KGTQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

And I looked into product image on bigreetech github and unlike Witbot one their picture does not show DIAG connection on motherboard:

[image]

does your board have those pins or not? If they do not then it certainly would not work, as second one is the DIAG pin.

b3nnb, 2020-04-28:
ya mine has the pins installed @rglory same as https://images.app.goo.gl/PxSARPgibmferY1n9 but without the jumpers installed

rglory, 2020-04-28:
> ya mine has the pins installed @rglory same as https://images.app.goo.gl/PxSARPgibmferY1n9 but without the jumpers installed

OK then you need to measure voltage on DIAG pin - 
[image]

of course with TMC2209 installed. I marked them for X, for others the same way. It should show the same low voltage.

b3nnb, 2020-04-28:
ok so just measure the voltage POS on the DIAG of the driver and NEG on the endstop of the matching axis?

rglory, 2020-04-28:
Just measure those voltage on those 2 points separately against ground. Ground can be on on middle pin on endstop, can be minus on power, or even case of USB connector should work - it's the same ground. Btw sorry for silly question - you do not have endstops connected and the same time to those axes?

b3nnb, 2020-04-28:
lol, ya no worries. no mechanical endstops are connected. ill measure those when i get a chance

b3nnb, 2020-04-28:
@rglory i measured the voltage and im getting about 0.02-0.03 with my multimeter set to 20V.
and its the same on the endstop jumpers and the the DIAG pins

rglory, 2020-04-28:
What is output of M119?

b3nnb, 2020-04-28:
really? since my probe uses ZMIN, how would i change that to use something else so i can use the ZMIN for the homing?

b3nnb, 2020-04-28:
i thought DELTAs ALWAYS used the MAX endstops, if i disable the MIX pins i get an error compiling

rglory, 2020-04-28:
Ah right you have delta. The issue that SKR v1.3 is wired on hardware level that X stepper controller DIAG pin is wired to X_MIN pin when you install that jumper. I think you can override that though and make X_MAX to use pin which is normally used as X_MIN and so on. Or you can manually rewire using cables that came with TMC2209, remove those jumpers and plug them. But I think that more messy. You can try this and see if it compiles:

#define X_MAX_PIN X_DIAG_PIN
#define Y_MAX_PIN Y_DIAG_PIN
#define Z_MAX_PIN Z_DIAG_PIN

also you would need to move your probe out of hardware -z connector. If you move it to z+ you should add this:

#define Z_MIN_PIN P1_24

 Let me know if that helps.

b3nnb, 2020-04-28:
so uncommenting the XMIN YMIN cause homing to go int the wrong direction for about 10mm and stop.
then i tried flipping the axis direction and homing will go up about 10mm and stop still but the Z motion is flipped
side issue, i dont know what i did but suddenly my E Driver vref wont go above .26. i was getting a error from M122 about it. so if i cant find a way to fix it, ill have to use a tmc2208_standalone from my previous board

b3nnb, 2020-04-28:
@rglory are you talking about this part in the pins.h file

```
//
// Assign endstop pins for boards with only 3 connectors
//
#ifdef X_STOP_PIN
  #if X_HOME_DIR < 0
    #define X_MIN_PIN X_STOP_PIN
    #define X_MAX_PIN -1
  #else
    #define X_MIN_PIN -1
    #define X_MAX_PIN X_STOP_PIN
  #endif
#elif X_HOME_DIR < 0
  #define X_STOP_PIN X_MIN_PIN
#else
  #define X_STOP_PIN X_MAX_PIN
#endif

#ifdef Y_STOP_PIN
  #if Y_HOME_DIR < 0
    #define Y_MIN_PIN Y_STOP_PIN
    #define Y_MAX_PIN -1
  #else
    #define Y_MIN_PIN -1
    #define Y_MAX_PIN Y_STOP_PIN
  #endif
#elif Y_HOME_DIR < 0
  #define Y_STOP_PIN Y_MIN_PIN
#else
  #define Y_STOP_PIN Y_MAX_PIN
#endif

#ifdef Z_STOP_PIN
  #if Z_HOME_DIR < 0
    #define Z_MIN_PIN Z_STOP_PIN
    #define Z_MAX_PIN -1
  #else
    #define Z_MIN_PIN -1
    #define Z_MAX_PIN Z_STOP_PIN
  #endif
#elif Z_HOME_DIR < 0
  #define Z_STOP_PIN Z_MIN_PIN
#else
  #define Z_STOP_PIN Z_MAX_PIN
#endif

```

and change Z_MAX_PIN to Z_DIAG_PIN type thing?

closed 2020-04-28 by thinkyhead

rglory, 2020-04-28:
I dont think Vref is used when you configure drivers through UART. Anyway I suggest you check ,arlin configuration repository for example configs for delta printers on SKR v1.3. Most probably somebody already hit this problem. 

Yes you can override pins in that file but that is not clean way. Better in Configuration.h you probably can do this:

#undef Z_MAX_PIN
#define Z_MAX_PIN P1_25

and so on for other axis. So you would keep all your changes in one place, which will simplify upgrade of firmware later.

b3nnb, 2020-04-28:
Sorry, i only tried adjusting it after i saw the Error: LOW when i ran M122. For the configs, my configs came from the examples. once i moved from 1.1.9 to 2.0.x i figured i would start from scratch, and here i am, lol.

Thanks, ill give the wiring a try. i REALLLLLLLY WANT sensorless homing on this printer!

i wish there were more docs on getting this working with DELTAs. im still waiting for a response to see if they have any recommendations.
I tried a bare Marlin 2.0.x from scratch. i tried bigtreetechs marlin branch for this specific board.

rglory, 2020-04-28:
I could help you making it work online but unfortunatelly in this format it happens too slow.

So looking at source code it seems that when sensorless homing is enabled and homing direction is positive marlin already would use negative hardware endstops as max ones on SKR 1.3. So I suggest you start with debugging your enstops. Remove diag jumps. Start with X. Connect mechanical endstop to **-X** (left connecter where 1.29 pin) and validate that M119 status for **x_max** is changed when you trigger it. Actual state is not important. If it does not you need to dig deeper. If it does do the same for Y and Z. Note you cannot use -Z for probe, you will have to switch to another. But you want to make endstops work. Then you can reconfigure probe.

b3nnb, 2020-04-29:
I GOT IT WORKING!!
I just decided to start from scratch on everything.
other then setting the board and the drivers and #define delta, i did what you mentioned first. enabling the neg endstops.
then enabled sensorless homing. this is all stuff i did before but i think i was more scattered with how i got there since its my first marlin 2.0.0

rglory, 2020-04-29:
I am glad it works!

Cheers