MarlinFirmware/Marlin #17760: [BUG] (short description)

grantpeterwilliamson, 2020-04-27:
Hi,

### Bug Description
I have a ender 3 with a SKR mini e3 motherboard
I've recently changed from marlin 2.0.x Bugfix which i've been on for a good while now up to 2.0.5.3.

and everything is working well apart from baby stepping, well to be a little more clear, its there until the print starts, it will heat up bed and nozzle, still there, it will home and auto bed level, still there, it prints my nozzle clean line, still there.. finishes nozzle clean line and moves to the first location to start printing, and its GONE! and double click  for z-stepping is gone too, and the entire "tune" menu, "pause" menu and "stop" menu, all three are gone, its as if the print thinks its sitting idle, apart from its not, its printing, and apart from those 3 menus not there and no double click for z-stepping shortcut it looks as it should do..

 ive updated all my configs over to the new build, and apart from the odd re-wording of commands and descriptions its all good.

i've  re-checked both configuration and configuration _adv about 5 times with diffchecker.com because i feel like i'm going mad, but still no go. 

not sure if it matters but i'm using octoprint as my host, no problems with that before this so i don't think its that..?

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4542597/Marlin.zip)

### Steps to Reproduce

1. flash from 2.0.x bugfix to 2.0.5.3
2. start print
3. wait for it to get past the formalitys and essentially start printing gcode
4. check for "Stop" "Pause" "Tune" menus to disappear and for baby stepping double tap to stop showing
**Expected behavior:** to show my menus and baby stepping

**Actual behavior:** it dosent.. lol!

grantpeterwilliamson, 2020-05-01:
i seem to have found a temporary solution to the problem,

 i have changed a fair bit in the last few weeks, one of which was my slicer software.
i noticed the X Y and Z was flashing right after it had done my nozzle clean gcode and the menus disappeared as im sure that meant it lost its position, so i started looking at my slicers..

i used to use cura, but changed to idea maker,

 i did port all my settings and starting gcode in to idea maker but after lots of re-flashing, setting changing i decided to try slice with cura, and low and behold, no problems at all.

 so i look at the Gcode output between cura and ideamaker and find a m1001 and m1002 command ive never seen before in idea makers output.
 
a little bit of googling and it seems to be a command specific to there printers, so going in to the.

gcode ---->  post possessing ----> ADD ---->command type "remove all occurrences of source" -----> then put M1001 ----> OK, repeat for m1002

 i tried to reslice and print it now prints perfectly

i have my pause, stop and tune menus back while printing and most importantly my baby stepping!!

and digging through the firmware files in visual studio code under Gcode SD M1001.cpp i find

/**
 * M1001: Execute actions for SD print completion
 */

that sounds like it would have deffiently caused the problem!

boelle, 2020-06-20:
@grantpeterwilliamson is this still an issue?

closed 2020-06-20 by grantpeterwilliamson