MarlinFirmware/Marlin #17761: TOOLCHANGE_PARK on deltabots ignores delta geometry

pfjason, 2020-04-28:
### Bug Description

TOOLCHANGE_PARK on deltabots: When moving from printing position to parked position delta geometry is ignored.  In my tests with park position at (0, Y_MIN_POS+10) toolpath is arc shaped and travels either through the model or through the bed. Tested in bugfix-2.0 at 66676e6d, but has been a problem for as long as I've tried to use the feature.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4542677/Configuration.zip)

### Steps to Reproduce

1. Start a print on a deltabot with multiple extruders and a toolchange_park location set to (0,Y_MIN_POS+10)
2. Initiate a toolchange
3. Notice vertical arc shaped move (if your nozzle is high enough) or bed crash if not.

**Expected behavior:** 
Linear move should use delta geometry and move on a flat Z plane.
**Actual behavior:** [What actually happens]
Prints are destroyed, printers are damaged.

boelle, 2020-06-20:
@pfjason is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle