MarlinFirmware/Marlin #17762: [BUG] TOOLCHANGE_MIGRATION_FEATURE does not reset position after ZRAISE

pfjason, 2020-04-28:
### Bug Description

When using TOOLCHANGE_MIGRATION_FEATURE with TOOLCHANGE_ZRAISE > 0 to manually migrate during print, after the ZRAISE and toolchange happens, nozzle does not return to previous Z Height and prints in midair after toolchange.  After print is aborted the migrated extruder then tries to extrude more than 1000mm of filament at high speed.  Tested on bugfix-2.0 as of 66676e6. 

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4542695/Configuration.zip)

### Steps to Reproduce

1. Start a print
2. Select Migrate to E2 (or whatever next tool is)
3. Observe ZRAISE -> ToolChange -> Print in midair.
4. Cancel print
5. Observe printing tool try to extrude around 1000mm of filament at high speed.

**Expected behavior:** 
Tool should return to level of print. When canceled, extruder should not then try to fire out the whole spool through the hot nozzle.

**Actual behavior:** [What actually happens]
Print continues in midair. When canceled, extruder tries to fire out the whole spool through the hot nozzle.

studiodyne, 2020-04-29:
I'm really sorry , but DELTA config files are not my experience, you must try to find someone here with this type of printer and good knowledge

Tool migration = Tool change , do you have same behavior  by just make a Toolchange T0 \n T1 ?

pfjason, 2020-05-01:
@studiodyne This shouldn't be effected by delta geometry. My other open issue is about TOOLCHANGE_PARK using the wrong movement type and therefore not moving in a flat Z plane, but this one is just about ZRaise not resetting back to previous position after tool migration using the LCD menu, only when TOOLCHANGE_ZRAISE_BEFORE_RETRACT is disabled. Since figuring out that I could make it work by enabling TOOLCHANGE_ZRAISE_BEFORE_RETRACT I haven't tried testing a normal toolchange.

boelle, 2020-06-20:
@pfjason is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle