MarlinFirmware/Marlin #17764: [BUG] G12 for T1 will not go less then X0

maukcc, 2020-04-28:
### Bug Description

Our clean point is located at X-19 Y35. T0 cleans good, but T1 will not go less then X0

### My Configurations

#define X_MIN_POS -29
#define Y_MIN_POS -96

  #define HOTEND_OFFSET_X { 0.0, -8.0 } 
  #define HOTEND_OFFSET_Y { 0.0, 84.0 }

  #define NOZZLE_CLEAN_START_POINT { -28, 30, }
  #define NOZZLE_CLEAN_END_POINT   { -10, 40,  }
  #define NOZZLE_CLEAN_CIRCLE_MIDDLE { -19, 35, }

### Steps to Reproduce

Home all axis
set nozzle at X100 Y100
select T0
G12 P1 T2 S1
select T1
G12 P1 T2 S1

**Expected behavior:** [What you expect to happen]

Home all axis
set nozzle at X100 Y100
select T0
G12 P1 T2 S1 (cleans nozzle at brush)
select T1
G12 P1 T2 S1 (cleans nozzle at brush)

**Actual behavior:** [What actually happens]

Home all axis
set nozzle at X100 Y100
select T0
G12 P1 T2 S1 (cleans nozzle at brush)
select T1
G12 P1 T2 S1 (goes in the direction of the brush stops at X0, and goes back immediately(no strokes))

#### Additional Information

We can manually go to the location of the cleaning with both T0 and T1
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

maukcc, 2020-04-28:
We also think the #define HOTEND_OFFSET_X { 0.0, -8.0 } negative 8 value has something to do with it. As we can recall that a previous T1 extruder with a positive X value did work.

thinkyhead, 2020-04-29:
New code was added to `nozzle::clean` by @InsanityAutomation that checks the software endstops. However, it doesn't check `soft_endstops_enabled`. We'll need to add that check so you can use `M211` ahead of cleaning to get the soft endstops out of the way.

thinkyhead, 2020-05-01:
This has been fixed in #17788.

closed 2020-05-01 by thinkyhead

maukcc, 2020-05-04:
What is the method of operation for this? 
As it still does not seem to work.

thinkyhead, 2020-05-10:
You can use `M211 S0` to disable software endstops before `G12` for more freedom of movement.

maukcc, 2020-05-11:
Even with that M211 workaround, it does not seem to work