MarlinFirmware/Marlin #17769: [BUG] MKS SGEN L - axis X and Y moving away from endstop when homing Z

48design, 2020-04-28:
### Bug Description
I can move the steppers like they should, homing X and homing Y is working fine but when Z is going to home both other axis are moving to their maximum and crashing into the frame.

### My Configurations
- endstops and directions configured
- z-probe installed using the min_z plug

### Steps to Reproduce
Home Z

**Expected behavior:**
- homing X
- homing Y
- homing Z

**Actual behavior:** [What actually happens]
- homing X
- homing Y
- instead of homing Z both axis X and Y are running into the frame

ellensp, 2020-04-28:
you have Z_SAFE_HOMING enabled and your X and Y steps/mm are wrong so that instead of moving to the center of the bed it moves into the frame.

ellensp, 2020-04-28:
This is not a support forum.
These are generic questions not firmware bugs or feature requests

Please use one of the services listed in 
https://github.com/MarlinFirmware/Marlin/issues/14694

48design, 2020-04-28:
I'm sorry, you are absolutely right. Wrong place. The thing is: I tried to change steps/mm and I used the same settings and printer but with a different board (MKS GEN L, 8bit) and after switching the board and copying all the settings (beside the board specific ones) I get this strange failure. That's why I thought it must be firmware related.

ellensp, 2020-04-28:
Go to the discord chat server https://discord.gg/n5NJ59y  lots of people able to help with configuration issues.

closed 2020-06-20 by boelle