MarlinFirmware/Marlin #17770: [BUG] (short description)

jan-gerard, 2020-04-28:
### Bug Description
When probing less than the maximum number of points for bed leveling, (e.g. `G29 P3`, while I have defined `#define GRID_MAX_POINTS_X 5`), the display says still `Probing Point 1/25`, while it should say `Probing Point 1/9`. 

### My Configurations
```
#define MULTIPLE_PROBING 3
//#define EXTRA_PROBING    1
#define AUTO_BED_LEVELING_LINEAR
#define GRID_MAX_POINTS_X 5
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```

### Steps to Reproduce
1. Set `#define GRID_MAX_POINTS_X 5` in Configuration.h
2. Compile and upload firmware. 
3. Level bed with `G29 P2`.

**Expected behavior:**
Probing Point 1/4 ... Probing Point 2/4 ... Probing Point 3/4 ... Probing Point 4/4 ... done.

**Actual behavior:**
Probing Point 1/25 ... Probing Point 2/25 ... Probing Point 3/25 ... Probing Point 4/25 ... done.

#### Additional Information
I think that line 702 in G29.cpp should be changed from 
`ui.status_printf_P(0, PSTR(S_FMT " %i/%i"), GET_TEXT(MSG_PROBING_MESH), int(pt_index), int(GRID_MAX_POINTS));`
to 
`ui.status_printf_P(0, PSTR(S_FMT " %i/%i"), GET_TEXT(MSG_PROBING_MESH), int(pt_index), int(abl_points));`

jan-gerard, 2020-04-28:
...and line 700 into 
`if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", int(abl_points), ".");`

jan-gerard, 2020-04-29:
I tested this change, and now it works perfectly.

boelle, 2020-06-20:
maybe submit a PR with the fix?

sjasonsmith, 2020-07-04:
@jan-gerard can you attach your configuration files, as required by the issue template? Ideally these would be updated to work with the bugfix-2.0.x branch.

There isn't enough information in your description for somebody to easily reproduce this. For example, it does not indicate which bed leveling mechanism you are using. Providing your configuration files will ensure that anybody looking at this is configured exactly the same way you are.

thinkyhead, 2020-07-05:
Thanks for the patch! It was a cinch to locate and to apply, and makes it in under the wire for 2.0.6….

Masanetz, 2020-07-05:
This patch seems to break compiling with AUTO_BED_LEVELING_BILINEAR, as abl_points nether declared nor has it a value...
`In file included from Marlin/src/gcode/bedlevel/abl/../../../inc/MarlinConfig.h:47:0,
                 from Marlin/src/gcode/bedlevel/abl/G29.cpp:27:
Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin/src/gcode/bedlevel/abl/G29.cpp:646:91: error: 'abl_points' was not declared in this scope
           if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
                                                                                           ^
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:111:57: note: in definition of macro '_SEP_2'
 #define _SEP_2(PRE,V)     serial_echopair_PGM(PSTR(PRE),V)
                                                         ^
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:174:45: note: in expansion of macro '_SELP_3'
 #define _SELP_5(a,b,V...)  do{ _SEP_2(a,b); _SELP_3(V); }while(0)
                                             ^~~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:168:28: note: in expansion of macro '_SELP_5'
 #define __SELP_N(N,V...)   _SELP_##N(V)
                            ^~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:169:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)
                            ^~~~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:201:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
                                 ^~~~~~~
Marlin/src/gcode/bedlevel/abl/G29.cpp:646:30: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
           if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
                              ^~~~~~~~~~~~~~~~~
Marlin/src/gcode/bedlevel/abl/G29.cpp:646:91: note: suggested alternative: 'abc_pos_t'
           if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
                                                                                           ^
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:111:57: note: in definition of macro '_SEP_2'
 #define _SEP_2(PRE,V)     serial_echopair_PGM(PSTR(PRE),V)
                                                         ^
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:174:45: note: in expansion of macro '_SELP_3'
 #define _SELP_5(a,b,V...)  do{ _SEP_2(a,b); _SELP_3(V); }while(0)
                                             ^~~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:168:28: note: in expansion of macro '_SELP_5'
 #define __SELP_N(N,V...)   _SELP_##N(V)
                            ^~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:169:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)
                            ^~~~~~~~
Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:201:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
                                 ^~~~~~~
Marlin/src/gcode/bedlevel/abl/G29.cpp:646:30: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
           if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
                              ^~~~~~~~~~~~~~~~~
*** [.pio/build/chitu_v5_gpio_init/src/src/gcode/bedlevel/abl/G29.cpp.o] Error 1`

sjasonsmith, 2020-07-05:
Re-opening, since the "fix" for this broke builds.

GhostlyCrowd, 2020-07-05:
Curious, i just did a fresh pull on latest bugfix enabled ABL Biliniar, and the bare minimum to get it to not throw errors (Z safe home and bltouch) on mega2060 and it compiles fine.

What platforms are failing and whats the confing files can you post them?

qwewer0, 2020-07-05:
On the latest bugfix, it throws error for me and for [MoneyMike](https://www.reddit.com/user/-MoneyMike-/) too.

```
'abl_points' was not declared in this scope
```

GhostlyCrowd, 2020-07-05:
> On the latest bugfix, it throws error for me and for [-MoneyMike-](https://www.reddit.com/user/-MoneyMike-/) too.
> 
> ```
> 'abl_points' was not declared in this scope
> ```
> 
> Edit: SKR Mini E3 v1.2 and v2.0

What are you using to compile VS code or Arduino IDE?

qwewer0, 2020-07-05:
I use VSCode

sjasonsmith, 2020-07-18:
@jan-gerard I believe this issue has been fixed. Can you re-test using `bugfix-2.0.x` and verify whether any issue remains?

qwewer0, 2020-07-18:
@sjasonsmith I has been fixed for me.

commit ea520e67 (2020-07-05): Fix "Probing mesh point" message

Closes #17770
closed 2020-07-20 by thinkyhead