MarlinFirmware/Marlin #17771: [BUG] (short description)

guffy1234, 2020-04-28:
### Bug Description
Hi

It seems that Marlin has an issue with X or Y probing to negative direction. For Z all is ok.

As example:

The sequence
G91
G38.3 X10
G90
Works properly.

But the sequence
G91
G38.3 X-10
G90
In most cases didn’t make second touch. Stop somewhere in the middle in a random position. Sometime it stops even without making first touch.
When the position is not zero, as example, X=30 - probing to negative direction (actually to X=20) has absolutely same issue

Same behavior for Y.

Disabling software endstops doesn’t help.

### My Configurations

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4545809/MarlinConfig.zip)

### Steps to Reproduce

1. Prepare probe.
2. Goto Custome menu
3. Press "Probe X-10"

**Expected behavior:** 
1. Head move to negative side, touch the probe. 
2. Head move to positive direction for retract lenght.
3. Head move to negative direction with double retract lenght.
4. Head stop when touch the probe. 

**Actual behavior:** 
For most cases on step 4 it stops in some intermediate point and doesn't touch the probe.
Some time it even doesn't perform first touch.  

#### Additional Information

Firmware - latest dual endstops mpcnc Marlin for ramps.
https://github.com/Allted/Marlin/tree/MPCNC_Ramps_T8_16T_LCD_32step_DualEndstop
 and Marlin release 2.0.5 merged on top of it.
I think pure Marlin has same issue.

I have added a lot of debug messages into g38.cpp and it seems that retract_mm and all destinations calculated properly. 
Also I didn't found any obvious issies in the Endstops::update().
So it looks like some external reason exist that inspire stop of movement.

boelle, 2020-06-20:
@guffy1234 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

vector76, 2020-06-23:
This sounds familiar, I think I've seen this issue in the not too distant past.  I'll check if it's still an issue and ask to reopen if I can reproduce it.

boelle, 2020-06-23:
@guffy1234 i see the confused emoji, but no comment?

boelle, 2020-06-23:
@vector76 @guffy1234 

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

its updated daily and sometimes several times each day, so could well be the issue is fixed

and be quick at replying here so we dont think you have left the issue or we think its fixed

vector76, 2020-06-23:
Ok.

closed 2020-06-29 by boelle