MarlinFirmware/Marlin #17773: [BUG] (short description)

Wdfitzpat007, 2020-04-28:
### Bug Description

When compiling in Arduino IDE, user is given the error:  
missing binary operator before token "(" 

If the line is commented out, the compiler proceeds fails at a random use of EITHER somewhere down the line. Most of which compile fine, but enough bug it out to make it very difficult to compile. Only happens in Config_Adv, or perhaps Adv compiles first.

Clearly either this is either a syntax or libraries issue. Please educate me, I saw no documentation that mentioned any similar issues except a 2019 April issue that was closed same day with no resolution.

It would be great if the next poor sap that had this issue stumbled on the thread instead of posting another ignorant bug report. Do not see it as humoring someone who cannot help themselves, but precluding future frivolous bug reports.

ellensp, 2020-04-28:
Attach your configuration files!

ellensp, 2020-04-28:
Attach error log.

Wdfitzpat007, 2020-04-28:
One moment

Wdfitzpat007, 2020-04-28:
Error log and all the files used in compile
[Package.zip](https://github.com/MarlinFirmware/Marlin/files/4547041/Package.zip)
[News script.txt](https://github.com/MarlinFirmware/Marlin/files/4547042/News.script.txt)

ellensp, 2020-04-28:
It looks like you have dropped a modern Configuration.h file into a really old version of Marlin.
This will not work.  It doesn't have the required macros like the ENABLED macro.

Wdfitzpat007, 2020-04-28:
I configured the config and config adv for the machine the old marlin was written for, but I was worried about the manufacturer hiding custom pins somewhere so I grabbed the old files too. What do I need to do to fix this? I can look things up on my own, so long as I know where to start

closed 2020-04-29 by Wdfitzpat007

Wdfitzpat007, 2020-04-29:
Solution: A version was found specially configured for my machine. Some modifications were made to the firmware to reflect the actual performance of the printer ( no the bed cannot reach 140C, it fails at 90C )