MarlinFirmware/Marlin #17774: Board Lerdge X  comiler Error

cjsoong, 2020-04-28:
Board Lerdge X compilation error:
Can someone provide the correct configuration.h & platformio.ini?
[image]

xC0000005, 2020-05-01:
I usually compile for lerdge in Arduino, then encrypt the firmware afterwards. The "correct" configuration.h depends on what printer you're putting it into.

cjsoong, 2020-05-01:
> I usually compile for lerdge in Arduino, then encrypt the firmware afterwards. The "correct" configuration.h depends on what printer you're putting it into.

Structure like Creality Ender-5.

Evg33, 2020-05-01:
@xC0000005 can you show #define SERIAL_PORT for Lerdge X ?

Evg33, 2020-05-08:
@xC0000005 
with  default_envs = ARMED and -DHAVE_HWSERIAL1  compilation ok.
-DVECT_TAB_OFFSET must be 0x10000 or 0x8000 ?
Can you show platformio.ini section for LerdgeX ?

cjsoong, 2020-05-09:
> @xC0000005
> with default_envs = ARMED and -DHAVE_HWSERIAL1 compilation ok.
> -DVECT_TAB_OFFSET must be 0x10000 or 0x8000 ?
> Can you show platformio.ini section for LerdgeX ?
> 
> Encoded by https://cdn.hackaday.io/files/1625926956336128/lerdgedecrypt2.py and flashed by stock bootloader, but no info on serial (esp connector/usb connector).
 @Evg33
Does TFT display work?  How to upload to

Evg33, 2020-05-11:
Don't need TFT. Need only serial or usb.

sjasonsmith, 2020-05-29:
@xC0000005, when you build for Lerdge in Arduino, which HAL are you using: STM32 or STM32_F4_F7?

xC0000005, 2020-05-29:
HAL STM32.

Evg33, 2020-05-29:
> > How to upload to Lerdge motherboard?

Place encoded bin file to USB or SD and upgrade from menu.
(  https://cdn.hackaday.io/files/1625926956336128/lerdgedecrypt2.py  )

boelle, 2020-06-21:
@cjsoong still an issue?

cjsoong, 2020-06-21:
@boelle Compilation is currently OK.

boelle, 2020-06-21:
ok, will then close as its fixed

if not we can always reopen

closed 2020-06-21 by boelle