MarlinFirmware/Marlin #17778: Compilation error when E0_AUTO_FAN_PIN is PWM with gcc 7.3.0

larsch, 2020-04-28:
Configuration_adv.h: 

```
#define E0_AUTO_FAN_PIN 6
#define EXTRUDER_AUTO_FAN_SPEED 192
```

Compiler: 7.3.0-atmel3.6.1-arduino5 (via latest arduino IDE)

Result:

```
SanityCheck.h:1416:7: error: non-constant condition for static assertion
       static_assert(GET_TIMER(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
```

Reason:

`GET_TIMER(E0_AUTO_FAN_PIN)` expands to `&(*(&OCR4AL)` which is a non-constant expression.

Trivial example generating the same compiler error: https://gcc.godbolt.org/z/QA8Eyn

Workaround: Remove static_assert, and everything works well, or set EXTRUDER_AUTO_FAN_SPEED to 255.

Full configuration files are here: https://github.com/larsch/MarlinAluUltimaker/tree/1.1.9-aluultimaker/Marlin

boelle, 2020-06-20:
@larsch is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle