MarlinFirmware/Marlin #17780: 2 independent heaters with mixing extruders

rgl1396, 2020-04-28:
Hello,

I am trying to use 3 mixing extruders and 2 heaters that can have independent temperatures. I am not able to figure out how to make this possible. I defined 2 heaters in Conditionals_LCD, but the heater does not turn on the heater on my board. 

Any help would be greatly appreciated!

Thank you,
rgl1396

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

Before filing an issue be sure to test the "bugfix" branches to see whether the issue has been resolved.

-->

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

MikeEitel, 2020-04-28:
I do the same with a trick.
The 2 first extruders use the same nozzles.
Patching the output of the heaters was not obviouse to do, but the heater feds short to ground. So you can easy conect the two pins, driving the heater cardridge. 
1. Marlin believes to control 3 heater/extruder combos. 
2. In code (pin file) i patch the second analog input and use here first pin too. 
Thats it. 
You program T0, T1, T2 and first two show always same temperature. 
Have fun

rgl1396, 2020-04-28:
So you set T0 and T1 to the same pin and T2 to another pin or the heater pins? I attached a pin assignment of the board I use. 

[image]

Thank you

MikeEitel, 2020-04-28:
Change pin 15 to pin 13 in the pin file of your board. 
That way your temp regulator of the first extruder and the reguator of the second have the same input value. If you ask for T0 first regulator will use his fet (honestly, i cant see with pin but gues it is one of the pwm pins 10?)
If you use T1 the second sw regulator will trigger his fet (pin 9?)
As you make a cable shortcut of the fet output your heater will be heated in both cases.
Hope i expressed myself clear enough. 
Have fun. Mike

chepo92, 2020-04-30:
>  
> 
> Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md
> 
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose
> 
> Before filing an issue be sure to test the "bugfix" branches to see whether the issue has been resolved.
> 
> 
> 
> ### Bug Description
> 
>   Description of the bug  
> 
> ### My Configurations
> 
> **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> 
> ### Steps to Reproduce
> 
> 
> 
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]
> 
> **Expected behavior:** [What you expect to happen]
> 
> **Actual behavior:** [What actually happens]
> 
> #### Additional Information
> 
> * Provide pictures or links to videos that clearly demonstrate the issue.
> * See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
> 

Please follow the template and **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

rgl1396, 2020-05-06:
Hello Mike,

I tried this and it showed the same temperatures on the thermistors, but did not allow me to set one temperature for T0 and another temperature for T1. 

**Expected behavior:** the ability to set T0 temperature to 200C and T1 temperature to 240C

**Actual Behavior:** when I set T0 to 200C it works but when I set T1 to 240C, it just sets T0 to 240C instead of setting its own temperature to 240C and just remains off.

I have attached my zip file that I have been using if you could take a look at it I would greatly appreciate it.

[GitHub Help.zip](https://github.com/MarlinFirmware/Marlin/files/4588168/GitHub.Help.zip)

Thank you,
rgl1396

MikeEitel, 2020-05-06:
Hi
Naturly not. 
T0 and T1 will have always the same temperature, also they heat via fet0 and fet2. Thats why you shortcut them and connect first heater cartridge. 
You have to use T2 and the fet2 for your second heater cartrige.
Is that understandable for you? 
Mike

MikeEitel, 2020-05-06:
Sorry typo: i meant fet0 and fet1

rgl1396, 2020-05-06:
What is fet(#) and where is it within marlin?

rgl1396

MikeEitel, 2020-05-06:
The current for a heating element is switched by an electronic element, called FET (use google if you need more knowledge). These are connected to bornes where u screw your cables of the heater cardriges. So if you have three of such high current heaters you connect the first two to the first cartridge. And the third to the second...
Mike

rgl1396, 2020-05-06:
What do you mean by shortcut? Sorry for all the questions.

Thanks
rgl1396

MikeEitel, 2020-05-06:
Sorry, I'm a little bit in question what is your background in electricity etc. And concerned that i might lead you into something dangerous. Do you have somebody near to you who could assist you with wiring etc. Please look for help. 

I'm talking that you have the two FET's work in parallel.
Connected. As both regulation loops (0 and 1) use the same input and the same output they work with the first heater cartridge. T0 with first extruder, T1 with second. T2 controls second cartridge plus extruder three. 
For the electric part, please, look for nearby assistance. Tell him as a hint that the fet's switch to ground. 
Mike

boelle, 2020-06-20:
@rgl1396 is this an issue or a question?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle