MarlinFirmware/Marlin #17783: [BUG] Issues compiling MKS SGEN with Dual Z Stepper

Azrael321, 2020-04-28:
### Bug Description

It is not possible to compile Marlin for a MKS SGEN (LPC1769) with Dual Z. I get this Errors during the compiling:

> Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
> Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
>Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
>Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
>In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:32,
>                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
>                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
>Marlin\src\HAL\LPC1768\../../core/../inc/../pins/pins.h:1342:8: error: #error "No E stepper plug left >for Z2!"
>1342 |       #error "No E stepper plug left for Z2!"
>     |        ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:40,
>                from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
>                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
>Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:580:4: error: #error "If >NUM_Z_STEPPER_DRIVERS is 2, you must define 
>stepper pins for Z2."
>  580 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
>      |    ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:32,
>                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
>Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:1342:8: error: #error "No E stepper plug left for Z2!"
> 1342 |       #error "No E stepper plug left for Z2!"
>      |        ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
>                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
>Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:580:4: error: #error "If NUM_Z_STEPPER_DRIVERS is >2, you must define stepper 
>pins for Z2."
>  580 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
>      |    ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:32,
>                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
>Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:1342:8: error: #error "No E stepper plug left for Z2!"
> 1342 |       #error "No E stepper plug left for Z2!"
>      |        ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
>                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
>Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:580:4: error: #error "If NUM_Z_STEPPER_DRIVERS is >2, you must define stepper 
>pins for Z2."
>  580 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
>      |    ^~~~~
>Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
>*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
>*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
>*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:32,
>                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
>Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:1342:8: error: #error "No E stepper plug left for Z2!"
> 1342 |       #error "No E stepper plug left for Z2!"
>      |        ^~~~~
>In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
>                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
>Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:580:4: error: #error "If NUM_Z_STEPPER_DRIVERS is >2, you must define stepper 
>pins for Z2."
>  580 |   #error "If NUM_Z_STEPPER_DRIVERS is 2, you must define stepper pins for Z2."
>      |    ^~~~~
>*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4548464/Configuration.zip)

### Steps to Reproduce

Use Visual Studio Code and PlatformIO. If i use Board MKS_SGEN_L or MKS_SBASE (LPC1768) compiling is succesfull.

**Expected behavior:** Compiling succesfull

**Actual behavior:** Compiling failed

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-28:
confirmed.
pins_MKS_SGEN.h is the issue
It has the lines lines
#undef E1_STEP_PIN
#undef E1_DIR_PIN
#undef E1_ENABLE_PIN
Comment these out by prefixing with //
The reason for this is the lack of free pins, and someone decided that X,Y,Z,E0 with uart way more important than and E1

This lead to issue 2
you have TMC2208's these need at least 1 port for UART,
the default are 
#define X_SERIAL_TX_PIN                  P4_29  // J8-2  <-- can't use when you have E1
#define X_SERIAL_RX_PIN                  P4_29  // J8-2  <-- can't use when you have E1   
#define Y_SERIAL_TX_PIN                  P2_08  // J8-3  <-- can't use when you have E1
#define Y_SERIAL_RX_PIN                  P2_08  // J8-3  <-- can't use when you have E1
#define Z_SERIAL_TX_PIN                  P2_11  // J8-4  
#define Z_SERIAL_RX_PIN                  P2_11  // J8-4
#define E0_SERIAL_TX_PIN                 P2_13  // J8-5  <-- can't use when you have E1
#define E0_SERIAL_RX_PIN                 P2_13  // J8-5  <-- can't use when you have E1
#define E1_SERIAL_TX_PIN                not_set
#define E1_SERIAL_RX_PIN                 not_set

so you need to find free pins to use for naelry all the UART pins.

If you have a LCD, you cannot do this.. there are insufficient pins.
This may help you https://github.com/makerbase-mks/MKS-SGen/blob/master/Hardware/MKS%20SGEN%20V1.0_001/MKS%20SGEN%20V1.0_001%20PIN.pdf

ellensp, 2020-04-29:
This is fixed, updated file can be found here https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/lpc1769/pins_MKS_SGEN.h

Azrael321, 2020-04-29:
Now everything is fine, thx!

commit 36adbfa4 (2020-04-29): Fix MKS S-Gen standard pins (#17786)

Fixes #17783
closed 2020-04-29 by Azrael321