MarlinFirmware/Marlin #17784: MERLIN 2.0.xx no autohome?

geekgamer2001, 2020-04-28:
i just updated and i can't find auto home or half the features i saw on 1.1.9 including auto home

geekgamer2001, 2020-04-28:
i see it in the code `  // Auto Home
  //
  GCODES_ITEM(MSG_AUTO_HOME, G28_STR);
  #if ENABLED(INDIVIDUAL_AXIS_HOMING_MENU)
    GCODES_ITEM(MSG_AUTO_HOME_X, PSTR("G28X"));
    GCODES_ITEM(MSG_AUTO_HOME_Y, PSTR("G28Y"));
    GCODES_ITEM(MSG_AUTO_HOME_Z, PSTR("G28Z"));
  #endif`

ellensp, 2020-04-28:
so enable INDIVIDUAL_AXIS_HOMING_MENU in Configuration.h

geekgamer2001, 2020-04-29:
but I just want to home all

ellensp, 2020-04-29:
add the following line, after  GCODES_ITEM(MSG_AUTO_HOME_Z, PSTR("G28Z"));
 GCODES_ITEM(MSG_AUTO_HOME_ALL, PSTR("G28"));

geekgamer2001, 2020-04-29:
and cooldown just saw that was missing

zroadhouse, 2020-04-29:
I'm seeing similar disappearing menu items after rebasing my config onto the most recent bugfix-2.0.x. Maybe Streamline menu item logic (#17664) changed something? I'm compiling now on the commit prior to that one to see.

ellensp, 2020-04-29:
Cooldown only appears when hot end is hot

geekgamer2001, 2020-04-29:
it was if i recall correctly

geekgamer2001, 2020-04-29:
yeah i went to select cooldown to move my printer was heated

zroadhouse, 2020-04-29:
Preheat PLA had also disappeared from the Temperature menu. Reverting bugfix-2.0.x to f709c565a1782111d464f1b0659d7b46a12a497d worked for me. Strongly suspect something in the #17664 change.

geekgamer2001, 2020-04-29:
ya just saw that how odd it's like many options are gone I try reverting to that soon

ellensp, 2020-04-29:
What lcd's are you guys using?  (shouldn't make any difference, but...)

Evg33, 2020-04-29:
And "Preheat PLA"
1st menu items lost in space ? :)

geekgamer2001, 2020-04-29:
that produces an error

thisiskeithb, 2020-04-29:
I can confirm that several items are missing from the menus after https://github.com/MarlinFirmware/Marlin/commit/4f003fc7a715830da7324da2c90f8411df3b1efc / https://github.com/MarlinFirmware/Marlin/pull/17664

xoniq, 2020-04-29:
Can confirm, just updated my printer to latest git version, I miss several items, but the first item I noticed, I wanted to home all, but its missing.

zroadhouse, 2020-04-29:
You can find my configuration here if its helpful => https://github.com/zroadhouse/Marlin/pull/1

geekgamer2001, 2020-04-29:
well shall i close this since we know the root cause or wait till its fixed

geekgamer2001, 2020-04-29:
in the meantime, I'm going to revert to  [https://github.com/MarlinFirmware/Marlin/commit/f709c565a1782111d464f1b0659d7b46a12a497d]
and write down all the menu options then go back to current and write down menu options for that and comment the missing menu items in #17664

firepower9966, 2020-04-29:
Also missing Autohome in LCD menu, individual homing is enabled.

xoniq, 2020-04-29:
Another thing I cannot find back: cool down. I had the printer hot to change filament, but I could not cool it down without going to temperature and set it to 0.

geekgamer2001, 2020-04-29:
yeah and cooldown is a important feature

geekgamer2001, 2020-04-29:
revert to 4f003fc for now as that has all options

thisiskeithb, 2020-04-29:
> well shall i close this since we know the root cause or wait till its fixed

Please leave this issue open so it can be tracked.

thisiskeithb, 2020-04-29:
The breaking changes were reverted in https://github.com/MarlinFirmware/Marlin/commit/e8b0796fc70d06f59cb217dbe11a2ad695243ad1, so this issue can be closed.

Evg33, 2020-04-29:
all ok

thisiskeithb, 2020-04-29:
Yes, I tested the changes as well and can confirm it's been fixed 🙂

closed 2020-04-30 by geekgamer2001

chepo92, 2020-04-30:
Lol I just synced yesterday and facing this issue, syncing again

Evg33, 2020-04-30:
@chepo92 fixed in "Revert menu_item_if ahead of refactor" https://github.com/MarlinFirmware/Marlin/commit/e8b0796fc70d06f59cb217dbe11a2ad695243ad1