MarlinFirmware/Marlin #17785: ERROR after homing with z_multi_endstops

Trash27, 2020-04-28:
### Bug Description
if using 3 or 4 Z_AXIS and enable Z_MULTI_ENDSTOPS i get an error after homing the Z-Axis.
All Z-Axis moves correctly home but then shows the Errormessage "ERROR:Pinter haltet. kill() called!" an the printer needs to restart.

### My Configurations
PrinterBoard: BTT GTR V1.0

**#define NUM_Z_STEPPER_DRIVERS 4**   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  **#define Z_MULTI_ENDSTOPS**
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif

thinkyhead, 2020-04-29:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thinkyhead, 2020-04-29:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

thinkyhead, 2020-04-29:
Please ZIP up and drop your configuration on your next reply, as specified in the issue template.

Trash27, 2020-04-29:
[marlin conifig](https://drive.google.com/open?id=1L4YziarHoUaGSomWQ8PMJL3qFQKzXNam)
[LOG.txt](https://drive.google.com/open?id=1ZYfdAV6nwyKZLI60W5K6V69eiyTps6ZY)

closed 2020-06-20 by boelle