MarlinFirmware/Marlin #17787: [BUG] M43 not reporting correctly pins defined as analog input GTM32 Pro

chepo92, 2020-04-29:
### Bug Description
Using, lastest bugfix 2.0, M43 is not reporting correctly  pins defined as analog input, and its value
I think it's similar to #14718 and #14706? 
For now it doesn't affect temp control or is a problem, but a To Do. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4549201/configs.zip)

### Steps to Reproduce

1. Compile in VScode for GTM32
2. Upload
3. Open Serial monitor and send M43
3. Optionaly send M104 Stt and then M43, no changes in input or analog input 

**Expected behavior:** [What you expect to happen]
Should be acording to: 
#define TEMP_BED_PIN       PC0
#define TEMP_0_PIN         PC1
#define TEMP_1_PIN         PC2
#define TEMP_2_PIN         PC3

**Actual behavior:** [What actually happens]

Got: 
PIN:  32  PC0  (A10)  <unused/unknown>                       Input  = 1                       
PIN:  33  PC1  (A11)  <unused/unknown>                       Input  = 1                       
PIN:  34  PC2  (A12)  <unused/unknown>   Analog in =     0   Input  = 0                       
PIN:  35  PC3  (A13)  <unused/unknown>   Analog in =     0   Input  = 0   

#### Additional Information
Testing in a GTM32 Pro VB 

maybe related to #14718 and #14706? 

[M43 output.txt](https://github.com/MarlinFirmware/Marlin/files/4549188/M43.output.txt)
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

chepo92, 2020-04-29:
Just saw it's a known limitation of current M43
https://github.com/MarlinFirmware/Marlin/pull/14072

boelle, 2020-06-20:
@chepo92 is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle