MarlinFirmware/Marlin #17791: [BUG] Compilation warning "unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]"

Exalik, 2020-04-29:
### Bug Description

Compilation of the Marlin-bugfix-2.0.x has a new warning.

### My Configurations

[Marlin-bugfix-2.0.x Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4550234/Marlin-bugfix-2.0.x.Configuration.zip)

### Steps to Reproduce

Download, configure, and build Marlin Bugfix-2.0.x

**Expected behavior:** the build should success without this warning

**Actual behavior:** the build success but there's a "new" warning

#### Additional Information
Visual Studio Code
Version: 1.44.2
Commit: ff915844119ce9485abfe8aa9076ec76b5300ddd
Date: 2020-04-16T17:50:03.709Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 5.3.0-46-generic

Exalik, 2020-05-01:
Hello, I still have this warning with the current Bugfix-2.0.x

Can somebody help me please ???

Exalik, 2020-05-01:
The warning concerns line 220 of queue.cpp

This line is a #pragma of GCC (#pragma GCC diagnostic ignored "-Wrestrict") which has been recently updated in GCC. I use the GCC version 9.0.3

boelle, 2020-06-20:
@Exalik is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle