MarlinFirmware/Marlin #17793: Power out recovery fails (Creality CR10 V2 board)

CdRsKuLL, 2020-04-29:
Morning,

Currently using the latest 2.0.5.3 version as of this morning.  I have enabled the #define POWER_LOSS_RECOVERY in config_adv and enabled the PLR default to true. On the LCD menu Outage is set to ON.  When I power off the printer and power back on it starts as normal, no message to continue appears. I then look on the SD card and it does have a PLR file (attached). 

[PLR.zip](https://github.com/MarlinFirmware/Marlin/files/4550527/PLR.zip)

I don't think I've missed anything...  any help would be greatly appreciated.

Steve

chepo92, 2020-04-30:
> Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md
> 
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose
> 
> Before filing an issue be sure to test the "bugfix" branches to see whether the issue has been resolved.

> 
> ### Bug Description
> 
> Description of the bug
> 
> ### My Configurations
> 
> **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> 
> ### Steps to Reproduce
> 
> 
> 
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]
> 
> **Expected behavior:** [What you expect to happen]
> 
> **Actual behavior:** [What actually happens]
> 
> #### Additional Information
> 
> * Provide pictures or links to videos that clearly demonstrate the issue.
> * See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Please follow instrucions and template

TheMiguelBi, 2020-05-01:
> Morning,
>
> Currently using the latest 2.0.5.3 version as of this morning. I have enabled the #define POWER_LOSS_RECOVERY in config_adv and enabled the PLR default to true. On the LCD menu Outage is [...]

Which Display are you using? And from which SD do you print?

I have a very similar problem. Here you can read the result of the tests I performed today:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/644#issue-610793577

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle