MarlinFirmware/Marlin #17794: [BUG] ( the L6474 module cannot be enabled using the M17 instruction)

Celino-Lin, 2020-04-29:
### Bug Description

通过串口发送M17指令，电机并没有使能。

发送M18指令，电机可以去使能。

marlin 2.0.5.3 固件中没有添加L6474的使能指令。

```
/**
 * stepper/L64xx.h
 * Stepper driver indirection for L64XX drivers
 */
/**
 * Line 45
 **/ 

// Y Stepper
#if AXIS_IS_L64XX(Y)
  extern L64XX_CLASS(Y)         stepperY;
  #define Y_ENABLE_INIT()       NOOP
  #define Y_ENABLE_WRITE(STATE) (STATE ? NOOP : stepperY.free())  //这里有问题，Enable = NOOP
  #define Y_ENABLE_READ()       (stepperY.getStatus() & STATUS_HIZ)
  #if AXIS_DRIVER_TYPE_Y(L6474)
    #define Y_DIR_INIT()        SET_OUTPUT(Y_DIR_PIN)
    #define Y_DIR_WRITE(STATE)  L6474_DIR_WRITE(Y, STATE)
    #define Y_DIR_READ()        READ(Y_DIR_PIN)
  #else
    #define Y_DIR_INIT()        NOOP
    #define Y_DIR_WRITE(STATE)  L64XX_DIR_WRITE(Y, STATE)
    #define Y_DIR_READ()        (stepper##Y.getStatus() & STATUS_DIR);
    #if AXIS_DRIVER_TYPE_Y(L6470)
      #define DISABLE_STEPPER_Y() stepperY.free()
    #endif
  #endif
#endif
```
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

  如果因为LCD的功能，编译有报错，可以注释掉。原本的控制卡是不带有LCD部件的，我没有选择注释，是因为我需要这个。
````
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
````
[2.0.5.3(3DP001V1).zip](https://github.com/MarlinFirmware/Marlin/files/4550862/2.0.5.3.3DP001V1.zip)

### Steps to Reproduce

1. 上传程序到控制卡，串口可以接收到初始化信息；

2. 发送指令 “M17”，串口收到 “ok”，确认电机是否可以使能。（电机没有使能，依然可以自由旋转）

3. 发送指令 “G0 X10“，串口收到 "ok"，X轴电机运动到位置10.（电机不能自由转动）

4. 发送指令“M18” ，串口收到“ok”，判断电机是否可以去使能。（电机已经去使能，可以自由转动）

5.发送指令“M17”，串口接收“ok”，判断电机是否已经使能。(电机还是没有使能，电机可以自由转动)

**Expected behavior:** [What you expect to happen]
     发送M17指令后，电机不可以自由移动

**Actual behavior:** [What actually happens]
     发送M17指令，电机却可以自由移动

#### Additional Information
* Provide pictures or links to videos that clearly demonstrate the issue.

[image]

#### My Solution
  修正宏定义  #define X_ENABLE_WRITE(STATE)  
  原本定义 STATE ==1 时  X_ENABLE_WRITE == NOOP
  修改为 STATE ==1 时  X_ENABLE_WRITE ==  stepperX.hardStop()

  Original code
````
/**
 * stepper/L64xx.h
 * Stepper driver indirection for L64XX drivers
 */

Line 44 
// X Stepper
#if AXIS_IS_L64XX(X)
  extern L64XX_CLASS(X)         stepperX;

    #define X_ENABLE_INIT()       NOOP
    #define X_ENABLE_WRITE(STATE) (STATE ? NOOP : stepperX.free()) 
    #define X_ENABLE_READ()       (stepperX.getStatus() & STATUS_HIZ)

  #if AXIS_DRIVER_TYPE_X(L6474)
    #define X_DIR_INIT()        SET_OUTPUT(X_DIR_PIN)
    #define X_DIR_WRITE(STATE)  L6474_DIR_WRITE(X, STATE)
    #define X_DIR_READ()        READ(X_DIR_PIN)
  #else
    #define X_ENABLE_INIT()       NOOP
    #define X_ENABLE_WRITE(STATE) (STATE ? NOOP : stepperX.free())
    #define X_ENABLE_READ()       (stepperX.getStatus() & STATUS_HIZ)

    #define X_DIR_INIT()        NOOP
    #define X_DIR_WRITE(STATE)  L64XX_DIR_WRITE(X, STATE)
    #define X_DIR_READ()        (stepper##X.getStatus() & STATUS_DIR);
    #if AXIS_DRIVER_TYPE_X(L6470)
      #define DISABLE_STEPPER_X() stepperX.free()
    #endif
  #endif
#endif
````
My modified code ( Simple and crude, but I ’m not sure if it ’s appropriate)
````
// X Stepper
#if AXIS_IS_L64XX(X)
  extern L64XX_CLASS(X)         stepperX;

    #define X_ENABLE_INIT()       NOOP
    #if AXIS_DRIVER_TYPE_X(L6474)
      #define X_ENABLE_WRITE(STATE) (STATE ? stepperX.hardStop() : stepperX.free()) 
    #else
      #define X_ENABLE_WRITE(STATE) (STATE ? NOOP : stepperX.free()) 
    #endif
    #define X_ENABLE_READ()       (stepperX.getStatus() & STATUS_HIZ)

  #if AXIS_DRIVER_TYPE_X(L6474)
    #define X_DIR_INIT()        SET_OUTPUT(X_DIR_PIN)
    #define X_DIR_WRITE(STATE)  L6474_DIR_WRITE(X, STATE)
    #define X_DIR_READ()        READ(X_DIR_PIN)
  #else
    #define X_ENABLE_INIT()       NOOP
    #define X_ENABLE_WRITE(STATE) (STATE ? NOOP : stepperX.free())
    #define X_ENABLE_READ()       (stepperX.getStatus() & STATUS_HIZ)

    #define X_DIR_INIT()        NOOP
    #define X_DIR_WRITE(STATE)  L64XX_DIR_WRITE(X, STATE)
    #define X_DIR_READ()        (stepper##X.getStatus() & STATUS_DIR);
    #if AXIS_DRIVER_TYPE_X(L6470)
      #define DISABLE_STEPPER_X() stepperX.free()
    #endif
  #endif
#endif
````

[image]

_**I am a novice, writing this report for the first time, please forgive me if something is not done correctly.
Hope everyone can help each other**_

thinkyhead, 2020-04-29:
The most knowledgeable about L64XX steppers is @Bob-the-Kuhn, so we'll have to verify with him whether `hardStop` is a supported method to enable the steppers. However, it looks to me like it should be okay. I will apply the patch and if there's some issue then it can always be fixed up or reverted.

thinkyhead, 2020-04-29:
Ok, the patch has been applied in the `bugfix-2.0.x` branch and will be included in the next tagged release. Please let us know if you see any other issues.

Bob-the-Kuhn, 2020-04-30:
Yes, this code will send the correct command to turn on the stepper driver.

I don't see any harm in enabling the steppers via M17.  I just can't think of a reason that it would be useful.

This looks like a solution for a problem.  Perhaps there is a better solution.  What is the problem you're trying to fix?

Celino-Lin, 2020-05-08:
> 是的，此代码将发送正确的命令以打开步进驱动器。
> 
> 我认为通过M17启用步进器没有任何危害。我只是想不出有用的理由。
> 
> 这看起来像是解决问题的方法。也许有更好的解决方案。您要解决的问题是什么？

Thanks to Bob ’s reply, I applied marlin 2.0.5.3 to the STEVAL-3DP001V1 board, but it is not going well. The noise of the motor is very loud, and then I began to look for the cause of this problem.

I found that the marlin application library is "L6470", but the actual chip used by STEVAL-3DP001V1 is "L6474". Therefore, the macro definition corresponding to L6474 will be missing, which is why the **Enable** and **Disable** functions need to be replaced with **hardstop ()** and **free ().**

However, it was further found that the L6474 "ALARM_EN" and other registers (marked by the red box in the figure below) were not initialized in version 2.0.5.3. I am not sure whether these registers will have an impact? 

With reference to the STM32 official source code modified based on marlin 1.1.0, it is found that they have initialized these registers of L6474H .
https://github.com/St3dPrinter/Marlin4ST/blob/5766bde8457e2bc020ab938a7a2b9bbca470a44f/stm32_cube/STM32/Drivers/BSP/Components/l6474/l6474.c#L1272

[image]

So, I want to know if I need to configure these registers?
Looking forward to your reply, thank you very much.

Bob-the-Kuhn, 2020-05-08:
I'm pretty sure that the **ALARM_EN** register is a don't care.  

The other three registers can be used to optimize your system.  The defaults are usually OK.  I have not played with these values so I can not offer any guidance.

---

My experiences with mitigating motor noise:
- Mechanical vibration dampers are wonderful.  Here's a [NEMA 17 example](https://www.aliexpress.com/item/32904553948.html?spm=a2g0o.productlist.0.0.599810a7bpaefq&algo_pvid=313f1dd5-4cf1-46c5-8352-ebb67a7c85f2&algo_expid=313f1dd5-4cf1-46c5-8352-ebb67a7c85f2-2&btsid=0be3743615889453167795495ea036&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_).
- Increasing the number of micro steps helps.  Unfortunately the power needed to drive the stepper increases as number of micro steps increases.  You can get to the point where your system can not supply enough power.
- Marginal settings can result in skipped steps which can cause a lot of noise. Try increasing the power to the steppers, reducing maximum feed rate, reducing maximum acceleration and reducing the number of micro steps.

---

I'm going to get on my soap box here.

I'm sorry you purchased the STEVAL-3DP001V1 board.  It's a feature rich board based on a poor stepper driver.  There are lower cost alternatives out there that have better performance and much better support.

I replaced my L64xx system with a Big Tree Tech SKR PRO 1.1 and TMC stepper drivers.  I went with TMC5160 stepper drivers because I'm running high power NEMA 23 steppers and they don't need heat sinks.

thinkyhead, 2020-05-10:
There are maybe 5 people on Planet Earth using L6474 steppers with Marlin, so we welcome the opportunity to do more testing.

Celino-Lin, 2020-05-14:
Thanks Bob-the-Kuhn and thinkyhead.
I found the cause of the noise, it should be caused by the motor itself, I replaced a new motor, its noise has been reduced a lot.
Of course, I also agree that L6474H is really poor. If I have sufficient funds in the future, I will try to use TMC5160, but currently I can only use L6474H.
thank you all, I will continue to test.

boelle, 2020-06-20:
@Celino-Lin still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle