MarlinFirmware/Marlin #17795: M117 replaces colon (':') by space (' ') on LCD screen in Marlin FW version 2.0.5.3

jopl, 2020-04-29:
### Bug Description

On LCD screen: every colon (':') inside the string following M117 command is replaced by space. I am not sure if this is right behaviour, but as the repetier server sends ETA time (ETE is similar) in format "HH:MM:SS Day XX", resulting string is "HH MM SS Day XX", which is unpleasant.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Go to web page of Repetier Server (ver. Free 0.93.1 in my case)
2. On Console page ("http://A.B.C.D:3344/#!/printer/sCube/console" in my case) type down in command line text "M117 ETE 14:37:17 Day 29" and press "Send" button
3. See LCD screen. It shows "ETE 14 37 17 Day 29", i.e. without colons

**Expected behavior:** [What you expect to happen]
"ETE 14:37:17 Day 29" on the LCD screen

**Actual behavior:** [What actually happens]
"ETE 14 37 17 Day 29" on the LCD screen, i.e. without colons

thisiskeithb, 2020-04-30:
Tried the latest `bugfix-2.0.x`? I just sent `M117 ETE 14:37:17 Day 29` through OctoPrint and it works:
[image]

jopl, 2020-04-30:
I found out the problem origin. Repetier server removes the colons from the command line starting with version 0.75.2 ( https://forum.repetier.com/discussion/2009/pi-0-75-1-extruder-temperature-reported-incorrect-and-extrusion-stopped ). But there is also written: "in firmware definition for the server it can be disabled".

ellensp, 2020-04-30:
So not a marlin issue. Can you close the job? thanks.

jopl, 2020-04-30:
OK, I'll close the job, but before that I would like to mention I found the solution:
Edit line in marlin.xml file by ssh client, i.e. write on command line in for example in putty:
sudo nano /usr/local/Repetier-Server/firmware/marlin.xml
Find string "ascii" (CTRL+W), it should be at line:
<com okAfterResend="false" protocol="ascii" allowColonInAscii="false"/>
Change the appropriate "false" for "true" at the end of line, i.e.:
<com okAfterResend="false" protocol="ascii" allowColonInAscii="true"/>
But it is still not tested, so be carefull ...

closed 2020-04-30 by jopl