MarlinFirmware/Marlin #17796: [BUG] (  MENU_ITEM_IF(1) not working)

studiodyne, 2020-04-29:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-04-29:
There’s no information in your report (please follow the template), but could it similar to what we’re seeing in https://github.com/MarlinFirmware/Marlin/issues/17784?

studiodyne, 2020-04-29:
I've made a very short issue description , because , i know that the author has just merged the commit  it's just to inform him. all menus made with conditions are down
Thks

thisiskeithb, 2020-04-29:
> I've made a very short issue description , because , i know that the author has just merged the commit it's just to inform him. all menus made with conditions are down
> Thks

We have to guess at what you mean without more information. Please fill out the issue template in the future.

studiodyne, 2020-04-29:
It's ok , for the next time ,  the author have reversed the commit , may be , too much breaks on this macro

Thks

closed 2020-04-29 by studiodyne