MarlinFirmware/Marlin #17802: [BUG] Printer heats up wrong nozzle when printing a second time from SD card

axjurado, 2020-04-29:
### Bug Description

My machine has 2 nozzles, rumba+ board, custom build. When printing from SD first time (after reboot) the machine selects nozzle correctly, heats up and prints. No problem. The second time around the machine heats up the wrong nozzle (nozzle 2 in my case), and starts performing a simulated print (assume because nozzle 1 is cold). This failure is random and sometimes there is no problem with the second print. 
When this happens reset is required.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4554612/config.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Turn on or reset computer.
2. Print. No problem here.
3. Print second time right after the first one.

**Expected behavior:** [What you expect to happen]
Heat up the correct nozzle.

**Actual behavior:** [What actually happens]
Some times heats up the wrong nozzle - random issue.

#### Additional Information
I only print from SD card, so not sure if this happens when tethered to a computer.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-30:
got a small example .gcode file you could share?   Just in case needs particular gcode combination to trigger,

AnHardt, 2020-04-30:
> [T<index>] | Hotend index. If omitted, the currently active hotend will be used.

I guess this is not Marlins fault.
If the previous print ends with using second tool and the new one wants to use the first you have to use the the **T**-parameter or you have to select the tool before you start to heat. That may need a M400 to finish the tool-change before the M104 (when sent without T-parameter).

axjurado, 2020-04-30:
I have more time to play with this last night. It only happens with nozzle 1. If I am starting prints that use nozzle 2, they work fine.
So with nozzle 1:
Turn on the machine and print with nozzle 1. Prints fine. The second time printing with the same nozzle 1 - no other changes - (sometimes though) the printer heats up nozzle 2 and does not print because nozzle 1 is cold.
With nozzle 2, it works fine. Start with nozzle 2, and all the subsequent prints work just ine. Heats up the correct nozzle.

I was testing the linear factor K gcode (which is pretty small) which is generated at marlinfw.org - for reference

axjurado, 2020-04-30:
> got a small example .gcode file you could share? Just in case needs particular gcode combination to trigger,

ellensp, 2020-04-30:
There are no T0 or T1 commands in that gcode.

axjurado, 2020-04-30:
> There are no T0 or T1 commands in that gcode.

Understand. But still heats up nozzle 2 and does not print because nozzle 1 is cold. 

Maybe there should be somewhere in the code telling marlin that if the tool is not specified should be T0 ?

boelle, 2020-06-20:
@axjurado is this still an issue?

axjurado, 2020-06-20:
Yes. I have corrected making sure the tool is selected but seems like there is a fail safe missing in the software.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

axjurado, 2020-06-24:
I’ve been using the bug fix version from ~April 20th. Do I need to update it or is recent enough?
In any case the big is still there. 
I created a gcode today where printing from T1 and there is a purge line, but there is no T1 command before it, and does not print the purge line. There is a T1 command afterwards and then the print goes fine. I used S3D. 
Again this is intermittent. If I reset the machine and start with that code it works. The second time around is the issue.

sjasonsmith, 2020-06-25:
@axjurado please retest with the latest bugfix commits. A lot of change happens in bugfix in two months.

You will probably have to make changes to your configuration file to work with the latest bugfix.  If you still encounter the issue please upload new configuration files, to make it easier for any volunteers to help investigate the problem.

axjurado, 2020-07-25:
Sorry fo rate slow response. I have not had time to play with the machine and use a newer download. Hope to get to it this weekend. 

I recently installed a z probe on my machine and started using the bed leveling feature. It works fine with T0 and when using T1 the machine goes back to T0 during homing and bed leveling. I have part of my initial script to retract while leveling and then print a purge line before starting the print, and I have to insert a few T1 commends to make it work correctly otherwise goes back to extrude with T0 (which is prevented due to low temp). Seems like this is related to this issue where the machine goes back to T0 during homing and bed leveling (granted i am still suing a bit older set of files).

closed 2020-08-31 by github-actions[bot]