MarlinFirmware/Marlin #17804: [BUG] Building Marlin from 2.0.x  branch with TMC2208 using Makefile fails

hradec, 2020-04-30:
### Bug Description

So, I'm trying to build 2.0.x using the Makefile, and I keep getting a link error with "undefined reference", as follow: 
```
  CXX   src/libs/stopwatch.cpp
  CXX   src/libs/numtostr.cpp
  CXX   src/libs/nozzle.cpp
  CXX   src/libs/hex_print_routines.cpp
  CXX   src/libs/crc16.cpp
  CXX   src/libs/L64XX/L64XX_Marlin.cpp
  CXX   applet/Marlin.elf
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::beginSerial(unsigned long)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper11beginSerialEm+0xe): undefined reference to `SoftwareSerial::begin(long)'
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper11beginSerialEm+0x1a): undefined reference to `SoftwareSerial::stopListening()'
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::TMC2208Stepper(unsigned int, unsigned int, float, unsigned char)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208StepperC2Ejjfh+0xb4): undefined reference to `operator new(unsigned int)'
TMC2208Stepper.cpp:(.text._ZN14TMC2208StepperC2Ejjfh+0xc0): undefined reference to `SoftwareSerial::SoftwareSerial(unsigned char, unsigned char, bool)'
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::write(unsigned char, unsigned long)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper5writeEhm+0xc0): undefined reference to `SSwitch::active()'
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::read(unsigned char)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper4readEh+0x60): undefined reference to `SoftwareSerial::listen()'
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper4readEh+0x8a): undefined reference to `SoftwareSerial::stopListening()'
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper4readEh+0x14a): undefined reference to `SSwitch::active()'
collect2: error: ld returned 1 exit status
 ```

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4555606/config.zip)

### Steps to Reproduce

Just run "make" with my configuration files. 

**Expected behavior:** 
The build to finish successfully.

**Actual behavior:** [What actually happens]
It fails the build when building Marlin.elf

#### Additional Information
I'm using Arduino 1.8.2 downloaded from Arduino website. I tried on booth armv71 and x86_64 platforms and got the same result.

hradec, 2020-04-30:
I was able to "fix" the "SoftwareSerial" undefined references by adding   
```

ifeq ($(TMC), 1)
  LIB_CXXSRC += TMCStepper.cpp COOLCONF.cpp DRV_STATUS.cpp IHOLD_IRUN.cpp CHOPCONF.cpp GCONF.cpp PWMCONF.cpp DRV_CONF.cpp DRVCONF.cpp DRVCTRL.cpp DRVSTATUS.cpp ENCMODE.cpp RAMP_STAT.cpp SGCSCONF.cpp SHORT_CONF.cpp SMARTEN.cpp SW_MODE.cpp SW_SPI.cpp TMC2130Stepper.cpp TMC2208Stepper.cpp TMC2209Stepper.cpp TMC2660Stepper.cpp TMC5130Stepper.cpp TMC5160Stepper.cpp
+  LIB_CXXSRC += SoftwareSerial.cpp
endif
```
to the Makefile. so it does seems like the Makefile should be adding this source but it isn't. 
After that change, my error message when down to just "undefined" in the TMC2208Stepper.cpp:
```
  CXX   src/libs/hex_print_routines.cpp
  CXX   src/libs/crc16.cpp
  CXX   src/libs/L64XX/L64XX_Marlin.cpp
  CXX   applet/Marlin.elf
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::TMC2208Stepper(unsigned int, unsigned int, float, unsigned char)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208StepperC2Ejjfh+0xb4): undefined reference to `operator new(unsigned int)'
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::write(unsigned char, unsigned long)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper5writeEhm+0xc2): undefined reference to `SSwitch::active()'
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::read(unsigned char)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208Stepper4readEh+0x146): undefined reference to `SSwitch::active()'
collect2: error: ld returned 1 exit status
make: *** [Makefile:878: applet/Marlin.elf] Error 1
```

hradec, 2020-04-30:
ok... I fixed another little bit by adding another missing file, this time from TMCStepper Library: 
```

ifeq ($(TMC), 1)
  LIB_CXXSRC += TMCStepper.cpp COOLCONF.cpp DRV_STATUS.cpp IHOLD_IRUN.cpp CHOPCONF.cpp GCONF.cpp PWMCONF.cpp DRV_CONF.cpp DRVCONF.cpp DRVCTRL.cpp DRVSTATUS.cpp ENCMODE.cpp RAMP_STAT.cpp SGCSCONF.cpp SHORT_CONF.cpp SMARTEN.cpp SW_MODE.cpp SW_SPI.cpp TMC2130Stepper.cpp TMC2208Stepper.cpp TMC2209Stepper.cpp TMC2660Stepper.cpp TMC5130Stepper.cpp TMC5160Stepper.cpp
+  LIB_CXXSRC += SoftwareSerial.cpp SERIAL_SWITCH.cpp
endif
```
SERIAL_SWITCH.cpp fixes the other couple of undefined in the middle of TMC2208Stepper.cpp source code, but there's still one left: 

```
CXX   src/libs/hex_print_routines.cpp
  CXX   src/libs/crc16.cpp
  CXX   src/libs/L64XX/L64XX_Marlin.cpp
  CXX   applet/Marlin.elf
applet/arduino/TMC2208Stepper.o: In function `TMC2208Stepper::TMC2208Stepper(unsigned int, unsigned int, float, unsigned char)':
TMC2208Stepper.cpp:(.text._ZN14TMC2208StepperC2Ejjfh+0xb4): undefined reference to `operator new(unsigned int)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:878: applet/Marlin.elf] Error 1
```

not sure what this `new(unsigned int)` is now... :(

hradec, 2020-05-01:
ok... I finally figured out the problem: 
```
ifeq ($(TMC), 1)
  LIB_CXXSRC += TMCStepper.cpp COOLCONF.cpp DRV_STATUS.cpp IHOLD_IRUN.cpp CHOPCONF.cpp GCONF.cpp PWMCONF.cpp DRV_CONF.cpp DRVCONF.cpp DRVCTRL.cpp DRVSTATUS.cpp ENCMODE.cpp RAMP_STAT.cpp SGCSCONF.cpp SHORT_CONF.cpp SMARTEN.cpp SW_MODE.cpp SW_SPI.cpp TMC2130Stepper.cpp TMC2208Stepper.cpp TMC2209Stepper.cpp TMC2660Stepper.cpp TMC5130Stepper.cpp TMC5160Stepper.cpp
+  LIB_CXXSRC += SoftwareSerial.cpp SERIAL_SWITCH.cpp new.cpp
endif
```
It seems we need to include arduinos `new.cpp` to create the object for the `new` c++ functionality used in TMCStepper library. I guess arduino IDE adds that by default, as well as SoftwareSerial.cpp.

Anyhow, the problem with the Makefile  when building for RAMPS with TMC2208, is the missing 3 .cpp files above. 

I'll make a pull request for this fix. 

But if someone runs into this type of problem in the future, the solution is documented here anyway.

marciot, 2020-05-29:
@hradec: The Makefile seems to have fallen out of favor in the Marlin community. I am the only one who maintains it for my own personal use and I don't use a TMC2208 :) But I do appreciate the effort you put into it and I do appreciate the PR.

boelle, 2020-06-21:
@hradec still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle