MarlinFirmware/Marlin #17807: [BUG] Touch Screen

kunpoolsion, 2020-04-30:
Hello.

Description

   Using a Fysetc S6 Board with Fysetc FT81050, touch screen stops responding after 7 seconds.

Steps to Reproduce

Enable ExtUI and set to FT81050. Fysetc S6 board PINMAP.

Expected behavior: Works properly

Actual behavior: TouchScreen stops responding after 7 seconds
Additional Information

Configuration.h:

//
// Touch UI for FTDI EVE (FT800/FT810) displays
// See Configuration_adv.h for all configuration options.
//
#define TOUCH_UI_FTDI_EVE

Configuration_adv.h
#if ENABLED(TOUCH_UI_FTDI_EVE)
// Display board used
//#define LCD_FTDI_VM800B35A // FTDI 3.5" with FT800 (320x240)
//#define LCD_4DSYSTEMS_4DLCD_FT843 // 4D Systems 4.3" (480x272)
//#define LCD_HAOYU_FT800CB // Haoyu with 4.3" or 5" (480x272)
//#define LCD_HAOYU_FT810CB // Haoyu with 5" (800x480)
//#define LCD_ALEPHOBJECTS_CLCD_UI // Aleph Objects Color LCD UI
#define LCD_FYSETC_TFT81050 // FYSETC with 5" (800x480)

// Correct the resolution if not using the stock TFT panel.
//#define TOUCH_UI_320x240
//#define TOUCH_UI_480x272
//#define TOUCH_UI_800x480

// Mappings for boards with a standard RepRapDiscount Display connector
//#define AO_EXP1_PINMAP // AlephObjects CLCD UI EXP1 mapping
//#define AO_EXP2_PINMAP // AlephObjects CLCD UI EXP2 mapping
//#define CR10_TFT_PINMAP // Rudolph Riedel's CR10 pin mapping
#define S6_TFT_PINMAP // FYSETC S6 pin mapping

When printer switch on, the touchscreen works properly for 7 seconds, then the screen goes back itself to information page and doesn´t respond to touches. If I reconnect the lcd to the board while still on it wotks for 7 seconds more.
Tried to disable sound and sd. Same behavior.
Thx

marciot, 2020-05-01:
Check the serial console to see if it is printing anything there.

marciot, 2020-05-01:
Also, support on the Fysetc is a totally new feature, being headed by @RudolphRiedel and others. Probably still bugs. Lot of discussion regarding this in https://github.com/MarlinFirmware/Marlin/issues/16628

P-C-R, 2020-05-01:
I have the same Problem

this is the debug

echo:Appending to DL from RAMG cache, bytes: 56 REG_CMD_DL: 56
Time to draw screen (ms): 28
echo:Returning to status due to menu timeout

kunpoolsion, 2020-05-01:
ups...
Newbie question: How to check the serial console?

kunpoolsion, 2020-05-01:
With latest version from Fysetcn(Master Branch) now it doesn´t freeze.
But anything you tocuh appears a fraction of a second and go back to Info screen, but Media that opens and let you navigate normally.

MESlammer, 2020-05-01:
I had same problem, It seems like there is something wrong in BaseScreen::onIdle() function inside base_screen.cpp, maybe variable `last_interaction` is not updating, and that causes screen to timeout forever:
`echo:Returning to status due to menu timeout`
`echo:New screen: 2`
`echo:Appending to DL from RAMG cache, bytes: 916 REG_CMD_DL: 916`
`Time to draw screen (ms): 28`
`echo:Appending to DL from RAMG cache, bytes: 916 REG_CMD_DL: 916`
`Time to draw screen (ms): 28`
`echo:Returning to status due to menu timeout`
Commenting whole part where IF check happens seems to fix this issue, but that's a janky workaround

> ups...
> Newbie question: How to check the serial console?
Do you use VS Code? There are Serial Monitor button at the bottom.

kunpoolsion, 2020-05-01:
thx,same errors:
`echo:New screen: 2
echo:Appending to DL from RAMG cache, bytes: 916 REG_CMD_DL: 916
Time to draw screen (ms): 27
echo:Appending to DL from RAMG cache, bytes: 916 REG_CMD_DL: 916
Time to draw screen (ms): 28
echo:Appending to DL from RAMG cache, bytes: 916 REG_CMD_DL: 916
Time to draw screen (ms): 28
echo:Returning to status due to menu timeout`

RudolphRiedel, 2020-05-01:
There are currently unresolved issues with modifications to a bunch of files.
On 2020-04-25 I had to use older versions of these files to get it running:
ftdi_eve_lib/compat.h
screens/base_screen.cpp
screens/bio_status_screen.cpp
screens/bio_tune_menu.cpp
screens/files_screen.cpp
screens/screens.cpp
screens/screens.h
screens/temperature_screen.cpp

And right now I can not even build but hopefully this is sorted out soon.

kunpoolsion, 2020-05-01:
Maybe I can try with the bugfix version on that date:
https://github.com/MarlinFirmware/Marlin/tree/f907de272aa8bbb5c9d25ae1473d395271e94044

kunpoolsion, 2020-05-01:
@RudolphRiedel Were you able to make it work in the past?

P-C-R, 2020-05-01:
I Tried this one

https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/S6/2.0.x

The s6 2.0.x branch NOT the s6 master.

There the Touch works. But no  tmc UART......

RudolphRiedel, 2020-05-01:
Yes, it did work just fine on my Cheetah running with 8MHz hardware-SPI.
https://user-images.githubusercontent.com/31180093/79631851-dbed7200-815b-11ea-9a1a-2049077fc07e.jpg

That is not the Fysetc FT81050 but also a 5" with 800x480, I designed the small board the Cheetah is connected to.

kunpoolsion, 2020-05-01:
@P-C-R Thx, even if I have tmc2209, will try just to see the screen working properly or try to merge with other branch

P-C-R, 2020-05-01:
Yes. But disable the tmc ones completly. Comment the drivers. ;)

kunpoolsion, 2020-05-01:
@RudolphRiedel Would you mind to share your branch?

marciot, 2020-05-01:
> I had same problem, It seems like there is something wrong in BaseScreen::onIdle() function inside base_screen.cpp, maybe variable `last_interaction` is not updating, and that causes screen to timeout forever

I just compiled with the timeout code and I was not able to reproduce that problem on my end. It seems to work fine for me. An easy workaround would be to set LCD_TIMEOUT_TO_STATUS to 0 or comment i out in `Configuration_adv.h`. If that works, then it definitely would be a  problem in the idle function.

marciot, 2020-05-01:
Incidentally, I noticed this touch UI code has suddenly become very active whereas before it was just crickets. What happened? Is it just that there is more free time right now to tinker with machines?

kunpoolsion, 2020-05-01:
Hahahahahahaha
True, "stay at home" rule makes everytone to focus on pending homeworks
Compiling your suggestion for try....

kunpoolsion, 2020-05-01:
@marciot Thanks, confirm that setting up timeout as 0 works perfectly.

P-C-R, 2020-05-02:
@marciot

I am not able to compile the offical 2.0.5.3.
Only If i Copy the LCD Folder from fysetc!

With the offical Bugfix its the Same Problem. Some SPI PIn is wrongly named
or so. But i think there is a PR Open for this!

closed 2020-05-02 by kunpoolsion

RudolphRiedel, 2020-05-02:
I put up a test-branch: https://github.com/RudolphRiedel/Marlin/tree/bugfix-2.0.x_extui_cheetah_test

This is of  course from here and not from the Fysetc Fork, these are mostly files from 2020-04-25 plus a couple of files from 2020-04-16.
This is setup to compile for and work with the Fysetc Cheetah and the FT81050.
I compiled it, have it up and running on my Cheetah V1.1a with my display.
Then I moved it to annother dir, disabled the few changes that are specific to my display, built it again without issues and did a git commit/push.
I have not printed with this setup, the board is just sitting on my desk, I copied over the configuration files for the Cheetah and enabled TOUCH_UI_FTDI_EVE.

> Incidentally, I noticed this touch UI code has suddenly become very active whereas before
> it was just crickets.
> What happened? Is it just that there is more free time right now to tinker with machines?

I was drawn into this from the sideline by calling out my name in the other thread. :-)
Not that I am stranger to Extui/EVE but that happened due to  my EVE library: https://github.com/RudolphRiedel/FT800-FT813
Well, I had the Cheetah lying around for months now and knew it could do hardware-SPI.
I also have quite a collection of different displays to play with. :-)
So the timing and the circumstances were right to get me involved with Extui/EVE again a little. :-)

kunpoolsion, 2020-05-02:
Thx for your work and time

kunpoolsion, 2020-05-02:
I was able to fork and compile the lastest bugfix commit just reverting the ofending files that produces the CLCD_SPI_BUS error (pins_BTT_SKR_MINI_E3.h and pins_BTT_SKR_E3_DIP.h) and making some changes in platform.io for Fysetc_S6 (my board) to ignore the software serial lib.
I will upload changes

kunpoolsion, 2020-05-02:
Finally used the fysetc lcd folder, tested and working with Fysetc S6 + Fysetc FT81050

P-C-R, 2020-05-02:
Yes the LCD timeout fixed IT. ;)

RudolphRiedel, 2020-05-02:
I checked it out and it does not build here.
And there is no display activated in Configuration.h. 

Well, that is older code that should not only be using software-SPI on the S6 but as software-SPI did not work on the Cheetah due to a timing issue it makes me wonder why it even works with the faster STM32F4.

So if this works, what code exactly is that using if not lcd/extui/lib/ftdi_eve_lib/ ?

kunpoolsion, 2020-05-02:
Is a mix between the Oficial bugfix and the lcd folder from fysetc, I just made some changes to fix some errors while compiling.
I left the config and config.adv as default, my config has some changes in pinouts and didn´t want someone has problems because of it.
I just fixed what I needed to make it work with S6, didn´t tried to compile for other boards.

RudolphRiedel, 2020-05-04:
I have the timeout problem as well now.

And commenting out LCD_TIMEOUT_TO_STATUS in Configuration_adv.h only means it is set later in Conditionals_post.h.
Setting LCD_TIMEOUT_TO_STATUS to 0 works for me as well.

base_screen.cpp:
void BaseScreen::reset_menu_timeout() {
  TERN_(LCD_TIMEOUT_TO_STATUS, last_interaction = millis());
}

Rolling this back to:
void BaseScreen::reset_menu_timeout() {
  #if LCD_TIMEOUT_TO_STATUS
    last_interaction = millis();
  #endif
}

And it works again with timeout.

kunpoolsion, 2020-05-04:
good one, thx

kunpoolsion, 2020-05-04:
Someone is able to have the Preheat button on screen?
I tried to enabled in config and I see the files in screens folders, but no way to make it shown.

marciot, 2020-05-05:
There are some preheat buttons in the change filament screen. They are not tied to the preheat settings in the config, however. The touch UI was designed for a specific brand of printer and isn’t a generic interface that exposes all the features in Marlin. This comes from how the development was funded, primarily. I don’t anticipate it ever being full featured UI.

kunpoolsion, 2020-05-06:
gosh! I think the change filamente menu is the only one didn´t check it XD
thx