MarlinFirmware/Marlin #17808: [FR] (feature request title)error compilation marlin 1.1.9 anet a2 plus ajout sdsupport

djbilbao, 2020-04-30:
hello everyone i installed marlin 1.1.9.1 on my anet a2 plus everything went well when i saw that there was no longer support for the SD card, so i was looking hard net and I found that we could activate the command by removing // define sdsupport my problem it is after that impossible to compile I have this error

Arduino : 1.8.12 (Windows 10), Carte : "Anet V1.0"

c:/users/djbil/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: C:\Users\djbil\AppData\Local\Temp\arduino_build_271856/Marlin.ino.elf section `.text' will not fit in region `text'

c:/users/djbil/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: region `text' overflowed by 10742 bytes

collect2.exe: error: ld returned 1 exit status

exit status 1
Erreur de compilation pour la carte Anet V1.0

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

ellensp, 2020-04-30:
This error is that the compiled firmware you just generated will not fit in the flash memory of your controller. You probably have a older 1284p based controller, it only has 128K flash...  You can spend months turning things on and off and finding the perfect combination for you that has acceptable number of features that will fit. Or you can upgrade to a controller with a larger flash size.   If you want to dor the former, take a look at https://www.youtube.com/watch?v=2TrGeL8ZdGU

djbilbao, 2020-04-30:
super thank you very much for the help i will try this good day cordially

ellensp, 2020-04-30:
Can you please close this issue then?

closed 2020-04-30 by djbilbao

ellensp, 2020-05-09:
Update:
many have has success with reducing the size by  
Enableing #define SLIM_LCD_MENUS in Configuration.h and disable //#define ARC_SUPPORT  In Configuration_adv.h