MarlinFirmware/Marlin #17809: [BUG] Re-ARM Thermal Runaway

psavva, 2020-04-30:
### Bug Description
Thermal Runaway is triggered with the Re-ARM Board.

I heated the hotend to 240 deg Celsius, and just let Repetier record the issue as it was happening.
Check the image below: 
[image]

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4559095/Marlin.zip)

Current revision of Marlin is: 015c4e6fbdd28f5c4cb5fc987d4f4ae513b1c13d

### Steps to Reproduce

Set the Target Tempreture of 240 Deg Celcius on the LCD for the Extruder and Wait.
Thermal Runaway is triggered (see graph above)

#### Additional Information

I had this issue happening with 2 different Thermistors.

GMagician, 2020-04-30:
Looking at hotend pwm, Marlin seems to react correctly. Have you checked the cartridge?

ellensp, 2020-04-30:
Seems to slowly cool down, my guess would be the thermistor is not tight on the hot end, or the fans are blowing on the hotend not the extruded plastic.

psavva, 2020-04-30:
This test was done with the fan off.

I will try with thermal paste and see if it makes a difference...

psavva, 2020-05-01:
Issue is the thermistor

closed 2020-05-01 by psavva