MarlinFirmware/Marlin #17810: [BUG] Defining U8GLIB_SH1106 leads to both HAS_SSD1306_OLED_I2C and HAS_SPI_LCD

alexeypetrenko, 2020-04-30:
### Bug Description

If you define `U8GLIB_SH1106` in `Configuration.h` both following flags are set:
 - `HAS_SSD1306_OLED_I2C`
 - `HAS_SPI_LCD`

### My Configurations

[My_config.zip](https://github.com/MarlinFirmware/Marlin/files/4559158/My_config.zip)

### Steps to Reproduce

 - Set `U8GLIB_SH1106` in `Configuration.h`
 - Add following assertion to the end of `./src/inc/Conditionals_LCD.h`:  
```
#if HAS_SPI_LCD == 1 && HAS_SSD1306_OLED_I2C == 1
  #error "WTF? HAS_SPI_LCD == 1 && HAS_SSD1306_OLED_I2C == 1"
#endif
```  
 - Build
 - See error being triggered
[image]

**Expected behavior:** No conflicting flags are set at the same time

**Actual behavior:** Two conflicting flags are setat the same time

#### Additional Information
##### Technical details
This happens due the folloging code path:
 - https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L205
 - https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L208
 - https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L344
 - https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L374

##### Why do I care?
I'm setting up I2C OLED display with SKR mini E3 v1.2, but following assertion fails: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3.h#L175 . It would not happen if `HAS_SPI_LCD` was not set.

alexeypetrenko, 2020-04-30:
Hacky solution that fixes this behavior:
Add following somewhere in the end of a file
```
#if HAS_SSD1306_OLED_I2C == 1
  #undef HAS_SPI_LCD
#endif
```

alexeypetrenko, 2020-04-30:
Hmmm. It does not compile if `HAS_SPI_LCD` is not defined.

So I guess, it is indeed needed. Although naming is quite confusing

sjasonsmith, 2020-05-02:
Would you be able to update your config for use in the current bugfix branch and post the updated version? I'm encountering far more issues with your config than what you are reporting here.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle