MarlinFirmware/Marlin #17811: Monoprice Mini Select v3/Pro license violation

Saturn49, 2020-04-30:
There has been other issues like this one raised in the past, but they have been closed:
https://github.com/MarlinFirmware/Marlin/issues/10552

I believe Monoprice is still in violation of the GPLv3 License for Marlin.  Nowhere in their documentation or on their website do they disclose how to get the firmware source for Marlin, which they are clearly using in their Mini Select (1, 2, and Pro/3) models.  Personally I have a Pro model so I wrote to Monoprice support and asked for the firmware.  This is the response I received:

"You can find that here: https://github.com/Monoprice3DP/MiniV2DeltaSourceCode"

Great, right?  Not really.  That's not the (full) source code to my printer, though it is a fork of Marlin.

The description of the source indicates it is for the Mini V1/V2 but not the Pro (sometimes called the V3). 

Additionally, there are differences between what's on my printer and what's in the source.  As a quick check, I looked at the M115 command.  The source code in Marlin/Core/language.h from the repo above indicates the result is:

"FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION " PROTOCOL_VERSION:" PROTOCOL_VERSION " MACHINE_TYPE:" MACHINE_NAME " EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) " UUID:" MACHINE_UUID

However, if I execute M115 on my printer I see:

NAME. Malyan    VER: 5.0        MODEL: M200     HW: HH02
CLK:EXT BUILD: Jul 13 2018 10:25:24
ok N0 P15 B15

I let them know this, and got this response:

"Sorry for the delay but the product manager was not able to provide us with the source code for this printer at the moment. The product manager is working to get it but I do not have an ETA so you would know when to expect that. Sorry about any inconvenience this may have caused. "

I will continue working with Monoprice on this, but I would like to at least keep this issue open until Monoprice provides a repo with the actual firmware they're using on their printers, not some random fork of Marlin.  I've asked if Monoprice plans to stop distributing the printers while they sort out their GPLv3 violation.

I don't know how these things usually go, but *I* am not the copyright owner for Marlin so there's nothing (legally) I can do other than ask, but the developers here may be able to do more, if they care to.

ellensp, 2020-04-30:
Are you going to shout us some lawyers? cause otherwise there is nothing that can be done.

ellensp, 2020-04-30:
Also this github is for marlin bugs and feature requests. Not your personal hunt for GPL violators.
Please close this issue and join the Marlin discord in the GPL section.

Saturn49, 2020-04-30:
Sorry if I struck a nerve.  I'm happy to continue this in Discord if that's the most appropriate place.  While I am not a lawyer, it may be possible to pressure (or embarrass) Monoprice into doing the right thing.  Also, since Amazon is a distributor of Monoprice 3d printers, that's another avenue worth pursuing.

closed 2020-04-30 by Saturn49