MarlinFirmware/Marlin #17812: [BUG] Line on center screen not working with CR10 display

p-r-o-m-e-t-h-e-u-s, 2020-04-30:
[image]

### Bug Description

One horizontal line in the middle of the screen is tuned off. This line does not work as soon as the power is turned on and the printer is booting.

Tested on branch `bugfix-2.0.x` from 2020-04-30 => Screen has issue
Tested on branch `bugfix-2.0.x` from 2020-04-30 => Screen work perfectly

### My Configurations
```
//
// Factory display for Creality CR-10
// https://www.aliexpress.com/item/32833148327.html
//
// This is RAMPS-compatible using a single 10-pin connector.
// (For CR-10 owners who want to replace the Melzi Creality board but retain the display)
//
#define CR10_STOCKDISPLAY
```
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/4560529/Archiv.zip)

### Steps to Reproduce

Just install lates 2.0.x branch. When I downgrade the firmware to my older version, it works again. 
So it's easily reproducible.

**Expected behavior:**
No non-working line on screen.

**Actual behavior:**
All lines are working on screen.

ellensp, 2020-05-01:
And there is the solid line along the top of your image.  
your description seem wrong. Same release date it works and doesn't work...?  Suspect a date is wrong.

p-r-o-m-e-t-h-e-u-s, 2020-05-03:
Yes, you are right. I've change one date. Thx.

boelle, 2020-06-20:
@p-r-o-m-e-t-h-e-u-s still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle