MarlinFirmware/Marlin #17813: [BUG] i2c eeprom hangs onm samd51

GMagician, 2020-04-30:
#17651 removed `eeprom_init` call from byte read/write functions but not to block counterpart (why removed partially?). This causes in SAMD51 a hang when accessing i2s eeprom.
I'll add code that currently is present only on lpc176x but not in all other hals...

May this lead to a more general problem?

p3p, 2020-04-30:
`eeprom_init` has been removed from the `eeprom_write_byte` function because you dont really want to reinitialise the i2c/spi peripheral on every byte,

The eeprom code is still being updated but it looks like currently the samd51 platforms new `eeprom_wired.cpp` has the `eeprom_init` call missing from `PersistentStore::access_start()`.

GMagician, 2020-04-30:
> eeprom_init has been removed from the eeprom_write_byte function because you dont really want to reinitialise the i2c/spi peripheral on every byte,

And why is it needed in block functions? I'm expecting it to be re moved even there

GMagician, 2020-04-30:
> eeprom_wired.cpp has the eeprom_init call missing from PersistentStore::access_start().

no platform but lpc1768 has suc call in access_start..

I'm fixing samd51 but wondering if other HALs have the same issue

p3p, 2020-04-30:
Thinkyhead is rewriting a lot of the eeprom access code across the platforms and there is currently a PR with further updates, I think it was just an oversight that removing `eeprom_init` from `eeprom_write_byte` would require calling `eeprom_init` in `PersistentStore::access_start()` for the PersistentStore implementations that use the old avr `eeprom` functions.

GMagician, 2020-04-30:
Posted a PR to fix

PR #17815 (2020-04-30): Fix samd51 i2c eeprom issues
When i2c eeprom is enabled samd51 simply hangs on startup. Now it's fixed

Fix #17813
closed 2020-04-30 by GMagician