MarlinFirmware/Marlin #17822: [BUG] Layershift every whole hour

mberntsen, 2020-05-01:
### Bug Description

Every time the print timer goes from xx:59 to xx:00 a layer shift appears in the print. My print takes a little over 6 hours and has 6 layer shifts.
I've printed two identical parts with layer shifts on the same height (when comparing visually).
I witnessed it happening when the print went from 4:59 to 5:00, a layer shift of a few mm appeared.
Shifts are not allways in the same direction, but heating before print start has a variable duration, so this might create a variation on the exact location in the file where the printer is at on the hour increase.

### My Configurations

zip file with config can follow later on request but my setup is as follows:
marlin version 2.0.5.3 (commit 0518dec60d0931745efa2812fa388f33d68cfa29)
Anycubic Predator, config files modified based on the Anycubic Kossel config file.
custom electronics with tmc2209 steppers, esp32, sd card and more on breadboard.
I2C oled display with U8g2 lib (yes, modified source to get this to work)

### Steps to Reproduce
start a print with duration > 1 hour

**Expected behavior:** [What you expect to happen]
no layer shifts

**Actual behavior:** [What actually happens]
layer shift on hour increase

#### Additional Information

o-nix, 2020-05-01:
I am currently suffering from random Y-axis **TMC2209** layers shifts, but I am not sure it is Marlin itself. Also, mine are not time-bound and I have EEPROM disabled.

mberntsen, 2020-05-01:
> I am currently suffering from random Y-axis **TMC2209** layers shifts, but I am not sure it is Marlin itself. Also, mine are not time-bound and I have EEPROM disabled.

How do you know it's random?

sjasonsmith, 2020-05-02:
@mberntsen I assume you have `PRINTCOUNTER` enabled? Try disabling it and see if the problem goes away.

mberntsen, 2020-05-02:
I've put two slashes before saveStats(); to disable it  and the print went fine tonight. when I've got time i'll try to debug the issue, see where the delay comes from within the eeprom saving..

Azrael321, 2020-05-03:
Since today, i get layershift, too. Yesterday is run a print with about 6h and everything was fine (Bugfix_20200502). Today i used the new Bugfix_20200503 and i get a layershift for about  in -Y after 1h of printing time. This layershift performs over 2 - 3 layers, this means i get a layershift in -Y and the next layer the same again. Before that layershift is can hear a strange "Crack" from somewhere i can not locate.

Grogyan, 2020-05-03:
This is likely a driver overheating issue.
I had this happen to me, and found that the drivers were intermittently shutting off briefly. 

Adding a fan to cool down the drivers eliminates this issue. Or reduce the driver current

GhostlyCrowd, 2020-05-04:
I'm in the same boat here.

I replaced the belts and y stepper, when replacing the y stepper i had to invert it as its a different stepper. and the layer shits changed direction from -y direction to +y shifting, with the axis inversion. I screwed around with it on the weekend and could not get it to stop layer shifting, its random but once it starts it just keeps on shifting like a stairway to print heaven.

sjasonsmith, 2020-05-10:
@mberntsen, can you post your configuration files? I would like to see which hardware you are using.

sjasonsmith, 2020-05-29:
@mberntsen are you still investigating this?

mberntsen, 2020-06-08:
I'm sorry, been busy, i'll try to upload is this week :)

boelle, 2020-06-21:
@mberntsen still an issue?

mberntsen, 2020-06-21:
these are my configurations.
I've still disabled the saveStats() function, and have not seen a single print defect ever since.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4809956/Configuration.zip)

closed 2020-08-29 by github-actions[bot]